qmk/qmk_firmware #13817: [Bug] jonavin's GMMK Pro layout RGB not working

tuschla, 2021-07-31:
I tried jonavin's GMMK Pro layout and it's working great except for everything backlight / RGB related. 
Do I have to use some other branch?

## System Information

 - Keyboard:  GMMK Pro
 - Operating system: Arch Linux
 - AVR GCC version: 11.1.0

 - ARM GCC version: 11.1.0

 - QMK Firmware version: 0.13.28

 - Any keyboard related software installed? 
   - No

zvecr, 2021-07-31:
Using `develop`?

tuschla, 2021-07-31:
Ok, using develop branch worked :)
But I had to copy /qmk_firmware/keyboards/gmmk/pro/keymaps/jonavin/rgb_matrix_map.h to /qmk_firmware/keyboards/gmmk/pro/ansi/keymaps/jonavin in order for it to work.

tuschla, 2021-07-31:
Also one or maybe more of the leds on the left side strip light up pink all the time
[image]

Jonavin, 2021-08-01:
The purple is the num lock not enabled indicator. If you want to change the behaviour to the opposite, remove the ! from this line 241 in keymap.c

  if (!IS_HOST_LED_ON(USB_LED_NUM_LOCK))

Jonavin, 2021-08-01:
You do have to compile against develop branch until they complete the branch merge on Aug 28 for the rgb to work.

timatoe, 2021-08-01:
Is there documentation on how to compile against the develop branch? 

Thank you.

hmfaysal, 2021-08-01:
Thanks @Jonavin for this branch, really loving the responsiveness of the keyboard. 

I have added NKRO support and decreased the debounce time to 0ms, as I do not need it. Also sending 12 keys per scan. Helped to achieve 1000hz scan rate at 1000hz polling rate.

Jonavin, 2021-08-01:
Thanks.

The develop branch also merged the rgb_matrix_map.h file into the wrong folder. The PR that fixed this has been merged into the develop branch. That should fix the issue a user reported on Discord.

closed 2021-08-02 by tuschla