qmk/qmk_firmware #13834: [Bug] Trying to compile for steno gives linking error(wheatfield/split75)

xsrvmy, 2021-08-01:
I am attempting to compile a steno layout for the wheatfield/split75 keyboard. When building, I get the linking error:
.../process_steno.o: In function send_steno_state:
.../process_steno.c:77: undefined reference to virtser_send

Is steno not supported for bootloadHID boards, or is something else wrong?

fauxpark, 2021-08-01:
Yes, the V-USB backend code does not have support for virtser - it probably is not possible to add support due to the endpoint limitations of V-USB, and because it leverages library code from LUFA (which is for the USB AVRs). THat stuff would have to be reimplemented by hand.

closed 2021-08-01 by xsrvmy

xsrvmy, 2021-08-02:
Would changing the bootloader do anything at all? Or is this limitation because of the chip?

BTW is there any documentation in QMK about how to check if a board will support a certain feature? When I chose boards in the future I probably want to make sure virtser is supported.

fauxpark, 2021-08-03:
The bootloader is a separate piece of code, unconnected to QMK. The limitation in this case is a combination of the MCU and the USB library QMK uses for this particular MCU. The ATmega32A does not have hardware USB capability, so it must be done completely in software using V-USB: https://www.obdev.at/products/vusb/index.html

Because of this, however, only USB low-speed is supported, and V-USB has a limited number of USB endpoints it can support (two interrupt/bulk-in, though our fork has squeezed in an extra one, and 7 interrupt/bulk-out).

Virtser requires one interrupt-in, one bulk-in, and one bulk-out endpoint (one for control signals, two for data).
One of the three available "in" endpoints is reserved for the keyboard interface, one is mainly for mediakeys/mouse, and the last one is usually given to Console, or Raw HID for VIA. So there is a bit of a tradeoff here.
Additionally, to my knowledge none of the backing code used to implement a CDC interface is present in the V-USB codebase.

xsrvmy, 2021-08-03:
Is the ATmega32A the only controller that have this limitation? Or are there other controllers with similar limitations?

fauxpark, 2021-08-03:
Anything that uses V-USB: https://docs.qmk.fm/#/compatible_microcontrollers