qmk/qmk_firmware #13852: Permission denied on when trying to flash with Arch package

mcp292, 2021-08-02:
After running into [this issue](https://github.com/qmk/qmk_firmware/issues/13602) I updated my `qmk` package and pulled the latest from `master. 

Now I get this error when trying to flash as a non-priveledged user:
```txt
make[1]: *** [tmk_core/rules.mk:432: .build/obj_crkbd_rev1_mcp292/cflags.txt] Error 1
```

The docs don't specify to use sudo and I've used this package without sudo for about a year.

Arch Linux
`qmk --version` 0.2.2
`git describe --abbrev=0 --tags` 0.12.29
crkbd

fauxpark, 2021-08-03:
Try `sudo make clean` first, then a normal build without sudo.

mcp292, 2021-08-03:
That seemed to work but I flashed and now am having trouble getting the other half of my keyboard to respond andthus go into bootloader mode. I suspect it's because I didnt notice this error:
```txt
Linking: .build/crkbd_rev1_mcp292.elf                                                               [OK]
Creating load file for flashing: .build/crkbd_rev1_mcp292.hex                                       [OK]
Copying crkbd_rev1_mcp292.hex to qmk_firmware folder                                               cp: cannot create regular file 'crkbd_rev1_mcp292.hex': Permission denied
```

fauxpark, 2021-08-03:
It seems the permissions on your qmk_firmware folder are messed up. You may want to reclone, but back up anything you need to keep (eg. keymaps) and make sure the permissions for those are correct too.

mcp292, 2021-08-03:
Did `chown -R user:user qmk_firmware` instead. (t is concerning that these got messed up. Would have also been a bummer if I couldn't short the pinholes on the corne as most of the default bootloader shortcuts required both keyboard halves to work.

Closing issue, but re-open if you see cause for concern.

closed 2021-08-03 by mcp292