qmk/qmk_firmware #13878: WSL2 - proper way to setup and switch branches?

pierrepgithub, 2021-08-04:
What is the proper way to setup WSL2 environment? I'm unsure if I'm doing it right. When I setup both branches master and develop, running qmk doctor shows this when I am on the develop branch:

```
$ qmk doctor
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 0.2.2
Ψ QMK home: /home/.../qmk/qmk_firmware
Ψ Detected Linux (WSL).
Ψ Git branch: develop
⚠ Git has unstashed/uncommitted changes.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.3.1
Ψ Found avr-gcc version 5.4.0
Ψ Found avrdude version 6.3-20171130
Ψ Found dfu-util version 0.9
Ψ Found dfu-programmer version 0.6.1
Ψ Submodules are up to date.
Ψ QMK is ready to go, but minor problems were found
```

```
$ git remote -v
origin  https://github.com/pierrepgithub/qmk_firmware.git (fetch)
origin  https://github.com/pierrepgithub/qmk_firmware.git (push)
upstream        https://github.com/qmk/qmk_firmware.git (fetch)
upstream        https://github.com/qmk/qmk_firmware.git (push)
```

I do the following:

1. git checkout master or develop
2. git fetch upstream
3. git pull upstream master or develop
4. git push origin master or develop

Is this wrong? I setup the fork using `git clone --recurse-submodules https://github.com/whoeveryouare/qmk_firmware.git`

Then I setup the qmk_home using `qmk setup <github_username>/qmk_firmware`

ironiridis, 2021-08-04:
What does `git status` say?

pierrepgithub, 2021-08-04:
```
┌──()-[~/qmk/qmk_firmware]
└─$ git status
On branch develop
Changes not staged for commit:
  (use "git add/rm <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
        deleted:    keyboards/gmmk/pro/ansi/keymaps/default/keymap.c

Untracked files:
  (use "git add <file>..." to include in what will be committed)
        keyboards/gmmk/pro/ansi/keymaps/pierrepgithub/

no changes added to commit (use "git add" and/or "git commit -a")

┌──()-[~/qmk/qmk_firmware]
└─$
```

Do I need to commit my keymap to my fork?

ironiridis, 2021-08-04:
The message from `qmk doctor` is warning you that you have untracked changes, which is true according to `git`. All that means is your files themselves are not in the git history. There's nothing to do here unless you're preparing to send your changes upstream to this repo.

Is something else not working?

pierrepgithub, 2021-08-04:
No, I was just curious on the warning.
It looks like I forgot to run git add . and commit in order to push my keymap to my develop fork
```
user@computer:~$ git add .
user@computer:~$ git commit -m "adding my keymap"
```

```
┌──()-[~/qmk/qmk_firmware]
└─$ git commit -m "adding my keymap"
[develop 354a035b1e] adding my keymap
 1 file changed, 12 insertions(+), 6 deletions(-)
 rename keyboards/gmmk/pro/ansi/keymaps/{default => pierrepgithub}/keymap.c (86%)

┌──()-[~/qmk/qmk_firmware]
└─$ git status
On branch develop
nothing to commit, working tree clean

┌──()-[~/qmk/qmk_firmware]
└─$ qmk doctor
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 0.2.2
Ψ QMK home: /home/pierre/qmk/qmk_firmware
Ψ Detected Linux (WSL).
Ψ Git branch: develop
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.3.1
Ψ Found avr-gcc version 5.4.0
Ψ Found avrdude version 6.3-20171130
Ψ Found dfu-util version 0.9
Ψ Found dfu-programmer version 0.6.1
Ψ Submodules are up to date.
Ψ QMK is ready to go

┌──()-[~/qmk/qmk_firmware]
└─$ git push origin develop
Username for 'https://github.com': pierrepgithub
Enumerating objects: 15, done.
Counting objects: 100% (15/15), done.
Delta compression using up to 16 threads
Compressing objects: 100% (8/8), done.
Writing objects: 100% (9/9), 2.30 KiB | 2.30 MiB/s, done.
Total 9 (delta 5), reused 0 (delta 0), pack-reused 0
remote: Resolving deltas: 100% (5/5), completed with 5 local objects.
To https://github.com/pierrepgithub/qmk_firmware.git
   6525b82f5f..354a035b1e  develop -> develop

┌──()-[~/qmk/qmk_firmware]
└─$
```

closed 2021-08-04 by pierrepgithub