qmk/qmk_firmware #13880: [Bug] tapping JS_BUTTON in Encoder Callback produces wrong key pressed

JulesRenz, 2021-08-04:
Hi, I'm currently working on my very first QMK powered Keyboard, which I'd like to setup as a Joystick device.
One of the features of this device will be three rotary encoders, that are supposed to press 6 (one each for CW and CCW rotation) of the 32 total Joystick buttons. While the remaining buttons work as expected, the buttons tapped in the `encoder_update_user` callback behave strange, see below.

Many many thanks for the effort you put into making QMK so amazing, so far I had a blast using it :)

## Describe the Bug

The tapping (via `tap_code16`) of JS_BUTTON1...6 produce the letters `ghijkl` instead.

This is the code in question:
```C
bool encoder_update_user(uint8_t index, bool clockwise) {
    dprintf("ENC:  %d, %d\n", index, (uint8_t) clockwise);
    if (index == 0) { /* First encoder */
        if (clockwise) {
            tap_code16(JS_BUTTON1);
        } else {
            tap_code16(JS_BUTTON2);
        }
    } else if (index == 1) { /* Second encoder */
        if (clockwise) {
            tap_code16(JS_BUTTON3);
        } else {
            tap_code16(JS_BUTTON4);
        }
    } else if (index == 2) { /* Second encoder */
        if (clockwise) {
            tap_code16(JS_BUTTON5);
        } else {
            tap_code16(JS_BUTTON6);
        }
    }
    return true;
}
```

The debug output shows the expected values for `index` and `clockwise`, however, you can see the keystrokes `ghijkl` typed uptop into to `Local file` textbox:
[image]

## System Information

 - Keyboard:  custom, Work in Progress
 - Operating system: OSX
 - AVR GCC version: 8.4.0

 - ARM GCC version: 8.3.1

 - QMK Firmware version: 0.13.29

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

fauxpark, 2021-08-04:
`tap_code()`/`tap_code16()` etc. only support [basic keycodes](https://docs.qmk.fm/#/keycodes_basic). I don't know if there is a way to trigger the Joystick keycodes without actually using the Joystick feature.

JulesRenz, 2021-08-04:
Hi, thanks for ansewring so quickly! 
OK, if that's the case it of course explains it. 

what exactly do you mean by "without actually using the Joystick feature"? The remaining buttons on the device are mapped to a `LAYOUT`, that sends the JS_BUTTONs like so:
```C
[_BASE] = LAYOUT(
                 JS_BUTTON2,  JS_BUTTON3,  JS_BUTTON4,  JS_BUTTON5 , JS_BUTTON0, \
    JS_BUTTON6,  JS_BUTTON7,  JS_BUTTON8,  JS_BUTTON9,  JS_BUTTON10 , JS_BUTTON1, \
    JS_BUTTON11, JS_BUTTON12, JS_BUTTON13, JS_BUTTON14, JS_BUTTON15, \
    JS_BUTTON16, JS_BUTTON17, JS_BUTTON18, JS_BUTTON19, JS_BUTTON20, \
    JS_BUTTON21, JS_BUTTON22, JS_BUTTON23, JS_BUTTON24, JS_BUTTON25  \
    )
```

is this what you mean by "using the Joystick feature"?

If not, how could I manually send JS_BUTTON presses from within the encoder callback?
Many many thanks in advance - cheers Jules

fauxpark, 2021-08-04:
At the moment, the code that handles keycodes is a bit of a mess, as QMK has grown organically out of TMK. As such, pretty much all "Quantum" keycodes (ie. keycodes that aren't in the Basic set or Basic with modifiers) can only be triggered by assigning them directly in the keymap.

Theoretically it should be perfectly possible to rework this so that you can trigger any keycode programmatically, however there is a bit of work to do first as some keycodes/features (like perhaps LED/RGB Matrix, with its reactive animations) require knowledge of the actual matrix position of the key being pressed, which obviously would not be present if you were calling eg. `tap_code(RGB_TOG)`.

JulesRenz, 2021-08-05:
Thanks for your thoughts on this!
OK, so I've looked into the code and it is indeed a bit complex, as you suggeested (and keep in mind, I'm not familiar with the codebase (yet?)).

I was able to work around the problem by writing a new function in my `keymap.c`
```C
#include "joystick.h"

void tap_joystickButton(uint8_t index)
{
    // press
    joystick_status.buttons[index / 8] |= 1 << (index % 8);
    send_joystick_packet(&joystick_status);

    wait_ms(20);

    // release
    joystick_status.buttons[index / 8] &= ~(1 << (index % 8));
    send_joystick_packet(&joystick_status);
}
```

and then calling to fromt the encoder callback:
```C
bool encoder_update_user(uint8_t index, bool clockwise) {
    dprintf("ENC:  %d, %d\n", index, (uint8_t) clockwise);
    if (index == 0) { /* First encoder */
        if (clockwise) {
            tap_joystickButton(26);
        } else {
            tap_joystickButton(27);
        }
    } else if (index == 1) { /* Second encoder */
        if (clockwise) {
            tap_joystickButton(28);
        } else {
            tap_joystickButton(29);
        }
    } else if (index == 2) { /* Second encoder */
        if (clockwise) {
            tap_joystickButton(30);
        } else {
            tap_joystickButton(31);
        }
    }
    return true;
}
```

While I suppose this is rather dirty, it serves my purpose for now.
Cheers
Julian

drashna, 2021-08-09:
closing, since resolved.

closed 2021-08-09 by drashna