qmk/qmk_firmware #13892: [Bug] Faulty inputs with recent windows 10 update

tsharky, 2021-08-05:
## Describe the Bug
With the recent Windows 10 update (weather thingy in the task bar) my Kinesis Advantage 2 keyboards with the [kinT keyboard controller](https://github.com/kinx-project/kint/) do not work properly on my working and private machines.
They worked flawlessly before the recent windows 10 update.

I noticed following issues:
- windows keyboard detection failed from time to time
- input is delayed/laggy
- spamming random keys strokes (like askl;hflasdhjk) will result in a freezed keyboard (no more inputs possible) and spamming the last input continuously (like askl;hflaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaaa.....). I need to unplug the keyboard to stop it.

On fedora or linux in general the keyboard works fine. I noticed it just on Windows and I cannot check it on MacOS.

## System Information

 - Keyboard:  kinesis/kint41 (not on master yet)
   - Revision (if applicable):  
 - Operating system: fedora 34
 - AVR GCC version: 10.2.0

 - ARM GCC version: 10.2.0

 - QMK Firmware version: 0.13.29

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context
@stapelberg told me to open an issue here. https://github.com/kinx-project/kint/issues/41

stapelberg, 2021-08-05:
> my Kinesis Advantage 2 keyboards

Because multiple keyboards are affected at the same time, I think a hardware defect is unlikely.

Apparently, even multiple different micro controllers (Teensy 3.6 on one keyboard, and Teensy 4.1 on the other) are affected.

Hence, I suspected a software compatibility issue between QMK and the latest Windows version.

tzarc, 2021-08-05:
I'm not running off a Teensy, but I'm definitely running ChibiOS-based boards... and I'm not seeing anything to that effect.
Which version of Windows? Maybe a `winver` is in order.

tsharky, 2021-08-06:
Hmm...
I am not 100 % sure what cause this issue..
Right now, I'm on Windows and it's working.

Before opening this issue and that whole day at work, it was not.
At work I did a windows update and after rebooting, I noticed these faulty inputs. So I thought I must have to do with the recent update.
Currently both keyboards work on both PCs.

My Windows versions are 20H2 (OS Build 19042.1110) and version 1909 (Build 18363.1679).

Maybe I reopen this issue if I have more details.

tsharky, 2021-08-11:
Worked fine until this morning.
Same behavior as described above.
I did several reboots, replugged the keyboard, ... No way to fix it.

Then I left it unplugged for 2-3 hours and now it works again.

My usual office keyboard is working normal.

tsharky, 2021-10-16:
Sorry, I can't tell you if it addresses #14814.
I only have basic knowledge in QMK and driver development.
But I have issues on Windows 10 only. The keyboard works fine on Fedora 34.

tsharky, 2021-12-22:
I guess, I found the solution for my issue.
I replaced the USB cable and till now, I don't have any issues on Windows.

closed 2021-12-22 by tsharky