qmk/qmk_firmware #13921: [Bug] X-Bows Knight Plus starts failing when firmware is close to size limit

Inventor22, 2021-08-08:
## Describe the Bug

After adding custom print code as triggered by a keypress, if the total firmware size exceeds 93% of total firmware capacity (atmega32u4), my keyboard will either stop showing RGB lighting effects and sporadically flash some or all keys white for an instant while typing.  If firmware size >= 94% of total firmware size available, the keyboard enters an infinite reboot loop immediately following firmware flashing, before any keys are pressed.

The firmware size is increased or decreased by adding or removing #DISABLE_RGB_MATRIX_* effects inside config.h

Rules.mk:
`EXTRAFLAGS += -flto
CONSOLE_ENABLE = yes`

Custom print code in qmk_firmware\keyboards\xbows\knight_plus\knight_plus.c:
`#define str(x) #x
#define xstr(x) str(x)

static const char *rgb_matrix_effects_string[] = {
    //xstr(RGB_MATRIX_NONE),

    #define RGB_MATRIX_EFFECT(name, ...) xstr(RGB_MATRIX_##name),
    #include "rgb_matrix_animations/rgb_matrix_effects.inc"
    #undef RGB_MATRIX_EFFECT

    //xstr(RGB_MATRIX_EFFECT_MAX)
};

static const size_t effectCount = sizeof(rgb_matrix_effects_string) / sizeof(rgb_matrix_effects_string[0]);

bool process_record_user(uint16_t keycode, keyrecord_t *record) {
  switch (keycode) {
    case KC_WWW_HOME:
      if (record->event.pressed) {
          print("hello world ");          
          uprintf("%d\n", effectCount);
          for (int effectIndex = 0; effectIndex < effectCount; effectIndex++) {
            uprintf("%s\n", rgb_matrix_effects_string[effectIndex]);
          }
      }
      return false;
    default:
      return true; // Process all other keycodes normally
  }
}`

## System Information

Keyboard: X-Bows Knight Plus
Operating system: Windows 10 Pro
AVR GCC version: avr-gcc.exe (GCC) 8.4.0
ARM GCC version: arm-none-eabi-gcc.exe (GCC) 10.1.0
QMK Firmware version: 0.10.51
Any keyboard related software installed? No

## Additional Context
Built with QMK commit ac2090bf4e8, which was committed on Sun Aug 8 00:45:43 2021 -0400

At 93% of firmware size, when RGB effects don't work but the keyboard is still responsive and types just fine, the following debug information appears to be printed to the HID output, as logged by QMK Toolbox:
`*** HID console connected: X-BOWS KNIGHT_PLUS (FEED:1227:0001)
  > keyboard_report: 
    r/c 0123456789ABCDEF
    00: 0000000000000000
    01: 0000000000000000
  > 02: 0000000000000000
    03: 0000000000000000
    04: 0000000000000000
    05: 0000000000000000
  > 
    r/c 0123456789ABCDEF
    00: 0000000000000000
    01: 0000000000000000
  > 02: 0000000000000000
    03: 0000000000000000
    04: 0000000000000000
    05: 0100000000000000
  > 
    r/c 0123456789ABCDEF
    00: 0000000000000000
    01: 0000000000000000
  > 02: 0000000000000000
    03: 0000000000000000
    04: 0000000000000000
    05: 0000000000000000
  > 
    r/c 0123456789ABCDEF
    00: 0000000000000000
    01: 0000000000000000
  > 02: 0000000000000000
    03: 0000000000000000
    04: 0000000000000000
    05: 0100000000000000
  > 
    r/c 0123456789ABCDEF
    00: 0000000000000000
    01: 0000000000000000
  > 02: 0000000000000000
    03: 0000000000000000
    04: 0000000000000000
    05: 0000000000000000
  > 
    r/c 0123456789ABCDEF
    00: 0000000000000000
    01: 0000000000000000
  > 02: 0000000000000000
    03: 0000000000000000
    04: 0000000000000000
    05: 0000000000100000
    mousekey [btn|x y v h](rep/acl): [00|0 0 0 0](0/0)
  > 
    r/c 0123456789ABCDEF
    00: 0000000000000000
    01: 0000000000000000
  > 02: 0100000000000000
    03: 0000000000000000
    04: 0000000000000000
    05: 0000000000100000
    rgb matrix mode [EEPROM]: 1`

And hitting the KC_WWW_HOME key prints the list of enabled effects to the console (captured through QMK toolbox as well), as intended:
`    RGB_MATRIX_SOLID_COLOR
    RGB_MATRIX_GRADIENT_UP_DOWN
    RGB_MATRIX_CYCLE_ALL
    RGB_MATRIX_CYCLE_UP_DOWN
  > RGB_MATRIX_CYCLE_OUT_IN
    RGB_MATRIX_CYCLE_OUT_IN_DUAL
    RGB_MATRIX_RAINBOW_BEACON
    RGB_MATRIX_RAINBOW_PINWHEELS
    RGB_MATRIX_RAINDROPS
    RGB_MATRIX_JELLYBEAN_RAINDROPS
  > RGB_MATRIX_TYPING_HEATMAP
    RGB_MATRIX_DIGITAL_RAIN
    RGB_MATRIX_SOLID_REACTIVE_SIMPLE
    RGB_MATRIX_SOLID_REACTIVE
    RGB_MATRIX_SOLID_REACTIVE_WIDE
    RGB_MATRIX_SOLID_REACTIVE_MULTIWIDE
  > RGB_MATRIX_SOLID_REACTIVE_CROSS
  > RGB_MATRIX_SOLID_REACTIVE_MULTICROSS
    RGB_MATRIX_SOLID_REACTIVE_NEXUS
    RGB_MATRIX_SOLID_REACTIVE_MULTINEXUS
    RGB_MATRIX_SPLASH`

drashna, 2021-08-09:
This is likely the chip running out of system memory, rather than program memory.  With a lot of features enabled, especially with some of the code that you've posted (eg, it looks like it's using strings in memory rather than progmem).

Inventor22, 2021-08-09:
Ah that's likely what's going on!  Didn't cross my mind to think of that.  I'll try storing those strings in flash rather than memory, and see if that fixes the problem.

Inventor22, 2021-08-09:
Yep, that was definitely the culprit. Below is the fix to put the pre-processor generated effect name strings in flash, and print them out.  Based off learnings from here: http://www.nongnu.org/avr-libc/user-manual/pgmspace.html
```
#define str(x) #x
#define xstr(x) str(x)

#define RGB_MATRIX_EFFECT(name, ...) const char STR_RGB_MATRIX_##name [] PROGMEM = xstr(RGB_MATRIX_##name);
#include "rgb_matrix_animations/rgb_matrix_effects.inc"
#undef RGB_MATRIX_EFFECT

PGM_P const rgb_matrix_effects_string[] PROGMEM = {
    //xstr(RGB_MATRIX_NONE),

    #define RGB_MATRIX_EFFECT(name, ...) STR_RGB_MATRIX_##name,
    #include "rgb_matrix_animations/rgb_matrix_effects.inc"
    #undef RGB_MATRIX_EFFECT

    //xstr(RGB_MATRIX_EFFECT_MAX)
};

static const size_t effectCount = sizeof(rgb_matrix_effects_string) / sizeof(rgb_matrix_effects_string[0]);
char effectNameBuffer[38];

bool process_record_user(uint16_t keycode, keyrecord_t *record) {
  switch (keycode) {
    case KC_WWW_HOME:
      if (record->event.pressed) {
          uprintf("Effect count: %d\n", effectCount);
          for (int effectIndex = 0; effectIndex < effectCount; effectIndex++) {
            strcpy_P(effectNameBuffer, (PGM_P)pgm_read_word(&(rgb_matrix_effects_string[effectIndex])));
            uprintf("%s\n", effectNameBuffer);
          }
      }
      return false;
    default:
      return true; // Process all other keycodes normally
  }
}
```

closed 2021-08-09 by Inventor22