qmk/qmk_firmware #13951: Omission in "How to Use GitHub with QMK" creates a tripping hazard

eric-s-raymond, 2021-08-10:
## Describe the Bug

## System Information

 - Keyboard:  
   - Revision (if applicable):  
 - Operating system:
 - AVR GCC version: 

 - ARM GCC version: 

 - QMK Firmware version:

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

eric-s-raymond, 2021-08-10:
Sorry, I fooed uop by typing enter in the title bar

The problem is that "How to Use GitHub with QMK" describes how to push changes to your fork but leaves out the crucial step of creating them on a local branch and pushing that branch.

Following the directions ass written would lead to an unhappy outcome. I noticed this and adapted, but a friend of mine who is a recent new contributor tripped over it.

I'd submit a PR to fix this directly, but I don't know Markdown well enough.

fauxpark, 2021-08-10:
Hm... this page actually needs a much more significant rework, since it recommends cloning the repo directly rather than installing the QMK CLI and running `qmk setup yourname/qmk_firmware`. Not to mention the rather outdated screenshots.