qmk/qmk_firmware #13956: [Feature Request] Changing debounce on the fly

andrebrait, 2021-08-10:
## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [x] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

Basically, Glorious has added a few key combos on the latest version of the stock firmware for the GMMK Pro, namely:
- Fn + D + 1 = 2ms
- Fn + D + 2 = 8ms
- Fn + D + 3 = 16ms (default, and that explains why my keyboard felt immediately a lot more responsive when I flashed QMK)

So, that got me thinking we could implement the same thing for QMK. On-the-fly changes for debounce time.

So I'm proposing the idea to enable that sort of functionality (up to each keyboard to decide whether or not to implement key combos for that, of course) and I can implement it, if that is ok.

I'm aware of the different algorithms and also some proposals to have a separate value for a downstroke debounce and an upstroke debounce such as https://github.com/qmk/qmk_firmware/issues/8763#issuecomment-622985070. That could be coordinated, of course.

A possible use-case is gaming and typing. While someone might prefer a longer time to ensure reliable typing, they might opt for a shorter time for gaming to ensure minimal latency.

So, what do you think?

drashna, 2021-08-11:
I'm not sure how much useful this would be, outside of initial testing.  And recompiling a few times for testing may be simpler, and will use less resources (a growing concern on AVR based boards).  

Additionally, the "custom" debounce type could be used to implement a variable debounce time, if you're so inclined.

andrebrait, 2021-08-11:
I believe I can probably find some C preprocessor magic that won't make that consume more resources on keyboards that don't use it (i.e. it will just use the existing DEBOUNCE macro).

I thought about using a custom one, but then I came to think that this could be useful to other people and it's something that can be used for the existing algorithms (easily for the symmetric ones, not that much for the asymmetric).

zvecr, 2021-08-13:
I too see limited value in the proposal.

> While someone might prefer a longer time to ensure reliable typing, they might opt for a shorter time for gaming to ensure minimal latency.

Switches don't suddenly become less bouncy because you launched "insert game". GMMK have to support various switches on a single firmware build, but the luxury of QMK is that you can recompile to get your desired behaviour.

The issue also extends out from just size constraints. Adding runtime configuration adds extra complexity which has to be maintained. The value would have to persist, so then you start getting user issues in the style of the classic "bootmagic keyswap", where they change the value without realising or without knowing how to reset to a known configuration. 

If we pivot the issue, is it that changing the value of debounce is too difficult? maybe due to it not being supported by something like config.qmk.fm? Because the longer term data driven efforts begin to solve that.

Hatbuster, 2021-08-18:
I don't see the point in this either, realistically. Industry standard is 5ms. So it's 5 recompiles at most.

It would make sense if the starting range is as high as glorious sets it.

But no idea why glorious would set the debounce delay to 16ms when every mx key switch I've seen is specced for 5ms or less.

andrebrait, 2021-08-30:
Ok. Got the point. Closing this.

Thanks for the input!

closed 2021-08-30 by andrebrait