qmk/qmk_firmware #13957: Combos documentation page should state clearly that dual-role keys are not supported

Evotron, 2021-08-10:
Combos documentation page should state clearly that dual-role keys are not supported. Currently, there is no information letting the user know that dual-role keys are not supported by combos.

I wasted an entire day trying to get combos to work because most of my keys are dual-role keys and I didn't know that was a limitation.

tzarc, 2021-08-10:
Try the `develop` branch - combos have been updated considerably.
If it still doesn't work, then we'd be [happy to accept a PR](https://docs.qmk.fm/#/contributing?id=how-do-i-make-a-contribution) to either fix the problem, or fix the docs!

Evotron, 2021-08-10:
I compiled my layout with the develop branch, and it indeed works!

closed 2021-08-10 by Evotron