qmk/qmk_firmware #13983: [Bug] Stack smashing/ empty elf

phentrophil100, 2021-08-12:
**This happens on any keyboard/keymap :**

Linking: .build/handwired_dactyl_manuform_5x6_default.elf                                           [ERRORS]
 | 
 | collect2: fatal error: ld terminated with signal 6 
 | compilation terminated.
 | *** stack smashing detected ***: terminated
 | 
gmake[1]: *** [tmk_core/rules.mk:344: .build/handwired_dactyl_manuform_5x6_default.elf] Error 1
Make finished with errors
gmake: *** [Makefile:539: handwired/dactyl_manuform/5x6:default] Error 1

Ψ QMK Doctor is checking your environment.
Ψ CLI version: 0.3.3
Ψ QMK home: /media/data/hardware/Keyboard/qmk_firmware
Ψ Detected Linux.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 8.3.1
Ψ Found avr-gcc version 5.4.0
Ψ Found avrdude version 6.3-20171130
Ψ Found dfu-util version 0.9
Ψ Found dfu-programmer version 0.6.1
Ψ Submodules are up to date.
Ψ QMK is ready to go

tzarc, 2021-08-12:
Seems environmental.
```
Linking: .build/handwired_dactyl_manuform_5x6_default.elf                                           [OK]
Creating load file for flashing: .build/handwired_dactyl_manuform_5x6_default.hex                   [OK]
Copying handwired_dactyl_manuform_5x6_default.hex to qmk_firmware folder                            [OK]
Checking file size of handwired_dactyl_manuform_5x6_default.hex                                     [OK]
 * The firmware size is fine - 20280/28672 (70%, 8392 bytes free)
```
```
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 0.3.3
Ψ QMK home: /home/nickb/qmk_build/qmk_firmware
Ψ Detected Linux.
Ψ CLI installed in virtualenv.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 7.3.1
Ψ Found avr-gcc version 5.4.0
Ψ Found avrdude version 6.3-20171130
Ψ Found dfu-util version 0.9
Ψ Found dfu-programmer version 0.6.1
Ψ Submodules are up to date.
Ψ QMK is ready to go
```

phentrophil100, 2021-08-13:
**dg to arm-none-eabi-gcc version 7.3.1
clear .build**
`Compiling: tmk_core/protocol/lufa/usb_util.c                                                        [OK]
Linking: .build/handwired_dactyl_manuform_5x6_default.elf                                           [ERRORS]
 | 
 | collect2: fatal error: ld terminated with signal 6
 | compilation terminated.
 | *** stack smashing detected ***: terminated
 | 
gmake[1]: *** [tmk_core/rules.mk:344: .build/handwired_dactyl_manuform_5x6_default.elf] Error 1
gmake: *** [Makefile:539: handwired/dactyl_manuform/5x6:default] Error 1
`
`Ψ Found qmk_firmware at /media/data/hardware/Keyboard/qmk_firmware.
Ψ QMK Doctor is checking your environment.
Ψ CLI version: 0.3.3
Ψ QMK home: /media/data/hardware/Keyboard/qmk_firmware
Ψ Detected Linux.
Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 7.3.1
Ψ Found avr-gcc version 5.4.0
Ψ Found avrdude version 6.3-20171130
Ψ Found dfu-util version 0.9
Ψ Found dfu-programmer version 0.6.1
Ψ Submodules are up to date.
Ψ QMK is ready to go
`

tzarc, 2022-08-29:
Honestly, no idea. All I'd suggest is to try a different machine. There's no issue building this on my local machine, nor on our CI platform.

closed 2022-08-29 by tzarc