qmk/qmk_firmware #14035: [Feature Request] System76 Launch Support

JSybrandt, 2021-08-17:
## Feature Request Type

- [ ] Core functionality
- [X] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

Allegedly, the firmware in the new System76 Launch keyboard is a fork of QMK. It would be great to see this as an available on the supported keyboards list.

Keyboard page on their website: https://system76.com/accessories/launch

tzarc, 2021-08-17:
They maintain [their own fork](https://github.com/system76/qmk_firmware).
Someone would need to attempt to upstream it to QMK, and so far nobody has done so.

andy0130tw, 2021-11-10:
Hello. I received my Launch Keyboard recently. Compiled system76's fork and it worked as intended. I wanted to have mouse keys but there seemed to be bugs around their maintained fork. So I tried to merge the latest tagged version of QMK (v0.14.29), and here it is: https://github.com/andy0130tw/qmk_firmware/tree/last-stable-dev

From the [comparsion](https://github.com/qmk/qmk_firmware/compare/master...system76:master#diff-8ec9cd07b7cfe13607d4cd7f8c749bba6db32940f2a50436ceac57e7615512fb) I can tell that there are only two modifications to the existing codebase, one has been already merged with PR #14286, and the other is a minor change to handle `atmel-dfu`. However, after I compiled and flashed my fork into my Launch Keyboard, the LED did not light at all. Keymap is fine.

I am new to this field and have never debugged QMK-related issue. Not knowing if there is a more suitable place to ask for help, either. Can anyone with Launch Keyboard help me test if it's working? Thanks.

closed 2022-01-11 by noroadsleft