qmk/qmk_firmware #14042: [Warning] Nested layout macro detected. Matrix data not available!

Lilalas, 2021-08-17:
Whenever I use the a command on my keyboard, e.g. `qmk info -kb planck/ez` or `qmk compile -kb planck/ez`, it will print this red warning in one of the first lines:

```
☒ keyboards/planck/ez/ez.h: LAYOUT_ortho_4x12: Nested layout macro detected. Matrix data not available!
```

However, the operation will then resume just fine and completes successfully, so I have always happily ignored this warning until now. But I decided I would like to find out what this warning means, what it is caused by, and maybe also how to suppress it if it’s not harmful.

I could not find any information about this warning in the documentation, apologies if I have missed anything! Do you have any idea?

## System Information

 - Keyboard: Planck EZ
 - Operating system: Ubuntu 20.04.2 LTS
 - AVR GCC version: `avr-gcc (GCC) 5.4.0`
 - ARM GCC version: `arm-none-eabi-gcc (15:9-2019-q4-0ubuntu1) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]`
 - QMK Firmware version: 0.13.33

drashna, 2021-08-17:
it's because of this:
https://github.com/qmk/qmk_firmware/blob/ab10ce49325c390644e3885fe479b87fe5879fd7/keyboards/planck/ez/ez.h#L43-L54

Lilalas, 2021-08-17:
As mentioned in @drashna’s pull request, a fix that has already been implemented in the development branch. Thank you!

closed 2021-08-17 by Lilalas