qmk/qmk_firmware #14058: [Bug] Raindrops, Jellybean Raindrops, and Digital Rain crash keyboard.

l33tlinuxh4x0r, 2021-08-18:
All other effects work correctly. Raindrops, Jellybean Raindrops, and Digital Rain cause the keyboard to lock up 100% have to unplug and replug to get the keyboard working again. Using latest source code to build qmk.

tzarc, 2021-08-18:
Need more info.

l33tlinuxh4x0r, 2021-08-18:
Whenever I change to any of the above listed effects My keyboard just dies... I am using a Redragon K556. I cloned the qmk firmware and then added only the k556 driver to it and compiled the firmware on archlinux. Everything else is vanilla and everything else is working correctly.

l33tlinuxh4x0r, 2021-08-18:
For what it is worth everything works fine in sonix qmk I just thought that I would update to the latest upstream before messing around further. I am thinking that since the only change that I made is the driver that this is probably an upstream issue.

HorrorTroll, 2021-08-18:
That bug seem like is from Custom matrix code problem from SonixQMK anyway, not from qmk itself. Need more details what cause for problem from SonixQMK anyway, QMK can’t help for that.

l33tlinuxh4x0r, 2021-08-18:
Thanks for the info.

l33tlinuxh4x0r, 2021-08-18:
Should this bug be closed then?

zvecr, 2021-08-18:
Yes, as we do not maintain that fork

closed 2021-08-18 by zvecr