qmk/qmk_firmware #14110: issue with compiling

pierrepgithub, 2021-08-22:
`$ qmk compile
Ψ Compiling keymap with gmake --jobs=1 gmmk/pro/ansi:pierrepgithub

QMK Firmware 0.13.36
Making gmmk/pro/ansi with keymap pierrepgithub

arm-none-eabi-gcc (15:8-2019-q3-1+b1) 8.3.1 20190703 (release) [gcc-8-branch revision 273027]
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Size before:
   text    data     bss     dec     hex filename
      0   38430       0   38430    961e .build/gmmk_pro_ansi_pierrepgithub.hex

gmake[1]: *** No rule to make target 'drivers/chibios/spi_master.c', needed by '.build/obj_gmmk_pro_ansi_pierrepgithub/spi_master.o'.  Stop.
Make finished with errors
gmake: *** [Makefile:539: gmmk/pro/ansi:pierrepgithub] Error 1`

`$ git status
On branch develop
nothing to commit, working tree clean`

tzarc, 2021-08-22:
`make clean` first?

pierrepgithub, 2021-08-22:
I ran `qmk clean`

Then qmk compile seemed to work after that

closed 2021-08-22 by tzarc