qmk/qmk_firmware #14127: [Bug] Keyboard stays lit when pc is in sleep, and will be unresponsive unless unplugged. DZ65RGB

ItsOzwald, 2021-08-23:
## Describe the Bug
I just built my first keyboard, so please excuse my lack of knowledge. After finishing the keyboard had two issues but managed to solve one of them. I had to re flash the stock firmware because I could not cycle through the RGB lighting effects, but I have yet to figure out the second problem. As the title says, whenever I place the PC in sleep mode, the keyboard stays lit, and the keyboard then becomes unresponsive. Pressing any of the keys will not wake up the PC, nor will any of the keys work after the PC is awake unless I unplug and re plug the keyboard. Any help would be greatly appreciated, thank you.    

## System Information

 - Keyboard:  
   - Revision (if applicable): KBDfans DZ64RGB V3 hot swap PCB  
 - Operating system: Windows 10 

 - QMK Firmware version: 0.1.1

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

vinorodrigues, 2021-08-24:
Possibly a duplicate of issue #13088

vinorodrigues, 2021-09-03:
This seems to be fixed now with pull request #14084, and then #14260

closed 2021-11-08 by drashna