qmk/qmk_firmware #14211: Tracking issue for keyboard removals

skullydazed, 2021-08-29:
This issue exists to track keyboards that we may want to remove from the repository.

There is currently no automation to nominate a keyboard, keyboards are added here when maintainers are faced with doing maintenance work on a board that seems to be abandoned or not worth the work it would take to bring it inline with current practices.

Current list-

* [ ]

skullydazed, 2021-08-29:
`handwired/reddot` throws this error:

```
☒ handwired/reddot: Non-assignment code found in rules.mk. Move it to post_rules.mk instead.
☒ Non-assignment code on line +11 keyboards/handwired/reddot/rules.mk: ifdef TEENSY2
☒ Non-assignment code on line +15 keyboards/handwired/reddot/rules.mk: else
☒ Non-assignment code on line +20 keyboards/handwired/reddot/rules.mk: endif
```

It has not had any changes, except by QMK Maintainers, since the original PR #1056.

@upils is this something you want to maintain in the repo?

upils, 2021-09-20:
@skullydazed As far as I know I am the only user of this layout. So I guess you may remove it if you want to do some cleanup. Thanks for your time maintaining and keeping the project clean :)

jackokring, 2021-11-22:
It seems that as everyone makes a fork to do their own keymap, then PR to build in that keymap making it all bigger. Would it perhaps be easier to automate a fork list page of modifications and or additions in the whole QMK ecosystem of a stable type?

In that way, a single index `*.md` file could be looked at for links to forked repositories with added keymaps. And maybe even a fetching script could be added to extend QMK like `qmk gitmaps <user-name>`.

I'm not sure how complex this would be to automate using github actions or some other thing.

tzarc, 2022-05-16:
Seems like we're done.

closed 2022-05-16 by tzarc