qmk/qmk_firmware #14242: [Bug] Preonic startup sound broken after firmware update

sbmueller, 2021-08-30:
After updating my fork of qmk_firmware after a long time and re-flashing, the startup sound of my preonic doesn't work anymore. Other sounds work.

## Describe the Bug

I have set in `rules.mk`:
```
AUDIO_ENABLE = yes
```
and in `config.h`:
```cpp
#ifdef AUDIO_ENABLE
    #define STARTUP_SONG SONG(PREONIC_SOUND)
    // #define STARTUP_SONG SONG(NO_SOUND)

    #define DEFAULT_LAYER_SONGS { SONG(QWERTY_SOUND), \
                                  SONG(COLEMAK_SOUND), \
                                  SONG(DVORAK_SOUND) \
                                }
#endif
```
In the past this was sufficient to get the startup sound when connecting the keyboard via USB. Now, I don't hear a sound. I can manually press the key for "AUD_ON" and get a audible feedback for that press, but startup sound is not enabled.

## System Information

 - Keyboard:  Preonic
   - Revision (if applicable):  rev3
 - Operating system: Mac OS and Ubuntu
 - AVR GCC version: 8.4.0

 - ARM GCC version: 8.3.1

 - QMK Firmware version: 0.14.3

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

I did not update my firmware since about spring this year. I did not need `AUDIO_ENABLE = yes` in my `rules.mk` back then, startup sound worked out of the box.

Cipulot, 2021-08-30:
Hi @sbmueller ,
I had the same problem with the same board but using QMK WSL. I ended up flashing [this](https://github.com/noroadsleft/qmk_bits/blob/master/eeprom_reset/preonic_rev3_default.bin) fimware to clear eeprom.
I then proceeded to recompile mine and flash it to my board.

hobby-grade, 2021-09-07:
I had this issue on a Planck, and while troubleshooting a different issue, [this seemed to solve the issue](https://github.com/qmk/qmk_firmware/issues/14067#issuecomment-904102352). Does this have any effect on your Preonic?

amadeus, 2021-09-07:
It does not appear to fix my planck, which has this issue (startup song still does not play)

Cipulot, 2021-09-07:
> It does not appear to fix my planck, which has this issue (startup song still does not play)

Same here, I have in both my prepnic rev3 and planck rev6. My bootmagic is set to `lite` on both

hobby-grade, 2021-09-07:
> > It does not appear to fix my planck, which has this issue (startup song still does not play)
> 
> Same here, in both my prepnic rev3 and planck rev6. My bootmagic is set to `lite` on both

Well in my solution the flag was set to "yes" and not "lite". Is this what you tried?

sbmueller, 2021-09-10:
I also can confirm I hear a sound occasionally without re-flashing anything. But most times, I don't.

Cipulot, 2021-09-10:
This isn't the first time that happens. Seems independent to which platform you're compiling on too...

Cipulot, 2021-11-03:
I've noticed the same in some occasions. Tho I think that my case has more to do with the establishment of a correct USB connection. I should do some more testing adding `WAIT_FOR_USB`.... I'll keep updated on my case

ReVanTis, 2021-11-04:
Hi @sbmueller @Cipulot ,

I guess a working host is required because the audio startup sound plays in matrix scan now, not right after audio init stage.

I've done a very simple test to connect preonic with a usb charger, and it actually fails to play the sound nor does the led lights up. It might relates to recent changes an refactoring audio and lighting system because I somehow remember both led and audio worked when connected to a simple charger quite a long time ago (might be 1-2 years ago).

Cipulot, 2021-11-04:
It might very well be that my usb hub acts a bit weird and therefore powers up the board without actually initializing the USB connection. I've had some cases in which USB hubs, especially during startup of the pc, seems to have a "delay" after power before initializing the USB connection. For that I'm going to directly connect the board to the pc so that I can eliminate this possible source of error.
Again I'll keep updated on the development...

JuanGarcia345, 2021-11-22:
I'm not sure that this may be of any help but it seems that the it was the commit [a3cbc8a004](https://github.com/qmk/qmk_firmware/commit/a3cbc8a004f6ec5b0e1df326353a2a2fc8221129) which broke the startup sound (tested with a preonic and a planck)

RobTheFiveNine, 2021-12-08:
> After some days I noticed that even with @ReVanTis 's fix my Preonic doesn't reliably play the startup sound. It seems to work _most_ of the time though.

I am noticing the same thing - tried reconnecting a few times to test and it sporadically doesn't work

drashna, 2021-12-24:
does `#define AUDIO_INIT_DELAY` help?

RobTheFiveNine, 2021-12-24:
> does `#define AUDIO_INIT_DELAY` help?

Yeah, it seems to work most of the time, there was a few times it didn't, but I've not been able to replicate it since originally commenting on here.

theamarin, 2022-12-02:
This problem still appears with my planck/rev5 using latest qmk versions.
On USB plug-in, at most a short "blirp" can be heard instead of the three-note startup sound when I use `#define AUDIO_INIT_DELAY`.
Without `AUDIO_INIT_DELAY` being set, no startup sound can be heard at all.

Also, in `quantum/keyboard.c:529`, I changed the line
```c
        } else if (timer_elapsed(delayed_task_timer) > 300) {
```
to
```c
        } else if (timer_elapsed(delayed_task_timer) > 3000) {
```
Now, it takes several seconds before the keyboard tries to play the startup sound, but it still sounds completely broken.
The timers and timestamps used for sound playback look good to me.
Any ideas where to debug this?

Apart from that, also the other sounds played back during e.g. `AU_ON` (audio on) and `AU_OFF` are somewhat scrambled, they are being played far too fast. To reduce this problem a bit, I specified in my keymap's `config.h` the following:
```c
#undef TEMPO_DEFAULT
#define TEMPO_DEFAULT 100
```
This not only reduces the speed by 20% as expected, but sounds like half the speed. But this is does not fix the startup sound and obviously is just a quirky workaround.
Do you think these issues might be related?

saiarcot895, 2023-06-19:
I have a Moonlander running a 1.5 month old QMK version, and without `AUDIO_INIT_DELAY`, I could hear only the first note of the startup sound. With `AUDIO_INIT_DELAY`, the full song plays.

I'm guessing it's the same or similar issue as to what's been reported here.