qmk/qmk_firmware #14246: [Bug]

andrebrait, 2021-08-30:
## Describe the Bug

The GMMK Pro is lacking the function layer on the online configurator. I've checked other keyboards that were added recently and they all show up just fine. I couldn't figure out why the GMMK Pro is only showing the keys assigned on the first layer.

I checked a number of information files and whatnot but everything seems to be the same for other keyboards that show up correctly (like the Keychron Q1).

## System Information

 - Keyboard:  GMMK Pro
   - Revision (if applicable):  
 - Operating system: Irrelevant
 - AVR GCC version: Irrelevant

 - ARM GCC version: Irrelevant

 - QMK Firmware version: Irrelevant

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

fauxpark, 2021-08-30:
[image]

andrebrait, 2021-08-31:
> [image]
> 

That's not at all what I'm talking about.

Switch to layer 1.
All keys are unassigned.
The default keymap has them assigned to a number of things (NKRO Toggle, RGB controls, multimedia keys, etc.), so that's what I would expect to see there. 

It seems to work fine for other keyboards but not the GMMK Pro.

fauxpark, 2021-08-31:
Then that would be an issue with the default keymap in the Configurator.