qmk/qmk_firmware #14262: Proposal of a new form of key matrix configuration.

mtei, 2021-09-01:
## Feature Request Type

- [x] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

The configuration of the current key matrix is done by defining one of the following three types.

* `#define DIRECT_PINS {...}`
* `#define DIODE_DIRECTION COL2ROW`
* `#define DIODE_DIRECTION ROW2COL`

For this configuration, `quantum/matrix.c` has the following structure.
This is difficult to read because of the long and complicated `ifdef ... endif` block.

```c
#ifdef DIRECT_PINS

//// direct_pins code block 27 lines

#elif defined(DIODE_DIRECTION)
#    if defined(MATRIX_ROW_PINS) && defined(MATRIX_COL_PINS)
#        if (DIODE_DIRECTION == COL2ROW)

//// COL2ROW code block 56 lines

#        elif (DIODE_DIRECTION == ROW2COL)

//// ROW2COL code block 58 lines

#        else
#            error DIODE_DIRECTION must be one of COL2ROW or ROW2COL!
#        endif
#    endif  // defined(MATRIX_ROW_PINS) && defined(MATRIX_COL_PINS)
#else
#    error DIODE_DIRECTION is not defined!
#endif
```

Therefore, I propose a new configuration format as follows.

* `#define KEY_MATRIX_TYPE DIRECT_SWITCH`
* `#define KEY_MATRIX_TYPE DIODE_COL2ROW`
* `#define KEY_MATRIX_TYPE DIODE_ROW2COL`

With this configuration, `quantum/matrix.c` will have the following structure, which is easier to read.

```c
#if KEY_MATRIX_TYPE == DIRECT_SWITCH
#   ifndef DIRECT_PINS
#       error "DIRECT_PINS is not defined."
#   endif
//// direct_pins code block 27 lines
#endif // KEY_MATRIX_TYPE == DIRECT_SWITCH

#if KEY_MATRIX_TYPE == DIODE_COL2ROW && defined(MATRIX_ROW_PINS) && defined(MATRIX_COL_PINS)
//// COL2ROW code block 56 lines
#endif // KEY_MATRIX_TYPE == DIODE_COL2ROW

#if KEY_MATRIX_TYPE == DIODE_ROW2COL && defined(MATRIX_ROW_PINS) && defined(MATRIX_COL_PINS)
//// ROW2COL code block 58 lines
#endif // KEY_MATRIX_TYPE == DIODE_ROW2COL
```

Also, add the following code to the top of `quantum/matrix.h` to ensure backward compatibility.

```c
#ifndef KEY_MATRIX_TYPE
// for backward compatibility
#    ifdef DIRECT_PINS
#        define KEY_MATRIX_TYPE DIRECT_SWITCH
#    else
#        if DIODE_DIRECTION == ROW2COL
#            define KEY_MATRIX_TYPE DIODE_ROW2COL
#        endif
#        if DIODE_DIRECTION == COL2ROW
#            define KEY_MATRIX_TYPE DIODE_COL2ROW
#        endif
#    endif
#    ifndef KEY_MATRIX_TYPE
#       error "KEY_MATRIX_TYPE could not be determined."
#    endif
#endif

#if KEY_MATRIX_TYPE != DIRECT_SWITCH && KEY_MATRIX_TYPE != DIODE_COL2ROW && KEY_MATRIX_TYPE != DIODE_ROW2COL
#    error "Invalid KEY_MATRIX_TYPE config."
#endif
```

tzarc, 2022-04-20:
Would it make more sense to do something like:
---
`rules.mk`:
```make
MATRIX_DRIVER = col2row
```
Then have `matrix_col2row.c` automatically included in the build, much like other drivers?
Can remove `DIODE_DIRECTION` and the like as a result... seems like it should allow for extensibility in the long run?

mtei, 2022-04-20:
> Would it make more sense to do something like:
> 
> `rules.mk`:
> 
> ```makefile
> MATRIX_DRIVER = col2row
> ```
> 
> ...then have `matrix_col2row.c` automatically included in the build, ...

I think that is a good approach too.

mtei, 2022-04-20:
On the other hand, we should also consider the fact that it is more maintainable not to split the file, as drashna has done by merging quantum/split_common/matrix.c into quantum/matrix.c.