qmk/qmk_firmware #14272: [Bug] BootloaderHID keyboard not recognized by my pixel 5 phone

xsrvmy, 2021-09-02:
I have noticed that my pixel 5 does not recognize my wheatfield/split75 as a keyboard. It does get recognized as a usb device, but the phone tries to charge it rather than considering it a keyboard. It doesn't even seem to be responsive - hitting the reset keycode does nothing. I can however enter bootloader by holding control when connecting the device.

The issue seems to be specifically with this combination of keyboard/controller/bootloader and my pixel 5 phone. Other Android 11 phones as well as iPads can detect the keyboard, and my phone can detect my new planck. The pixel 5 is known to be stricter than other phones with usb c cables, but this doesn't seem to be the problem here.

hsgw, 2021-09-15:
The MCU (ATMEGA32A) on the wheatfield/split75 does not support native USB.
So, it uses VUSB, which emulates the protocol by software, but unfortunately, it is not perfect and may be unstable and unrecognized by some hosts.

Try connecting it through a USB hub or try a shorter cable.
If it is still not recognized, it may have a hardware compatibility issue.

tzarc, 2022-08-29:
No updates in nearly a year. Feel free to reopen if/when more information comes to light.

closed 2022-08-29 by tzarc