qmk/qmk_firmware #14328: [Bug] rgb_matrix effect confusion after breaking changes of 2021Q3

moyi4681, 2021-09-06:
-keymap:
VIA
- Keyboard:  
DZ60RGB , DZ60RGB ANSI, DZ60RGBWKL
with source code after breaking change to compile a via firmware, the light effect is not correct
with release 0.13.38 works fine

kb-elmo, 2021-11-06:
Can confirm this.
It's not just an issue with DZ boards.

It's the link time optimization that causes this.
Enabling LTO completely breaks the RGB matrix now!

drashna, 2021-11-06:
Do you know what commit specifically breaks it?  

Because I haven't seen issues with rgb matrix and LTO being enabled.

kb-elmo, 2021-11-07:
This doesn't exactly look like "out of order RGB animation".
It just doesn't work anymore at all.

It works perfectly fine with LTO disabled (which is only possible with the non-VIA firmware because with VIA it would be too big for the flash)

[image]

drashna, 2021-11-08:
Could you try reverting 70fb3e1aaf406cbbd5137916a93ed814d6891ef2 and see if that fixes the issue?

kb-elmo, 2021-11-08:
> Could you try reverting [70fb3e1](https://github.com/qmk/qmk_firmware/commit/70fb3e1aaf406cbbd5137916a93ed814d6891ef2) and see if that fixes the issue?

yup reverting those changes fixed it.

RGB works now with LTO enabled.

sigprof, 2021-11-09:
With what versions of `avr-gcc` does the problem appear? E.g., Debian/Ubuntu distros have 5.4 with significant Atmel/Microchip modifications; MSYS uses 8.4.0; on Arch Linux the distro-provided version is usually rather recent (11.2.0 at the moment), but the QMK install script for Arch Linux tries to install the 8.3.0 version from archive. There might be significant differences between the LTO implementations in these versions.

kb-elmo, 2021-11-09:
> With what versions of `avr-gcc` does the problem appear? E.g., Debian/Ubuntu distros have 5.4 with significant Atmel/Microchip modifications; MSYS uses 8.4.0; on Arch Linux the distro-provided [...]

5.4.0 on Ubuntu WSL (in my case. No idea what version DZ uses)

Andrew-Fahmy, 2021-11-09:
I also had an issue with the rgb matrix changes breaking lighting on my K-type (specifically commit 70fb3e1aaf406cbbd5137916a93ed814d6891ef2). I removed `__flash` and it seemed to fix my issue but I'm not really sure if this is a great way to solve it

moyi4681, 2021-11-10:
looks like the issue appears happen on IS31FL3733 microchip. IS31FL3741 works fine, I've not tested on other microchip.

Andrew-Fahmy, 2021-11-12:
After more testing it seems like the problem was not at all related with the `__flash` and that it was just a coincidence that it worked when I changed that. I think it has something to do with flashing the firmware.

I build two versions of the firmware, one from master and the other from a older commit (70fb3e1aaf406cbbd5137916a93ed814d6891ef2). Both of them work fine, but if I flash the same firmware a second time the lighting doesn't work at all.

Could this be a bug with the bootloader?

kb-elmo, 2021-11-12:
> Could this be a bug with the bootloader?

Doesn't apply for me.

Only disabling LTO or removing the `__flash` fixes the issues here.

Andrew-Fahmy, 2021-11-13:
> Only disabling LTO or removing the `__flash` fixes the issues here.

Neither of these fixes my problem.  There is another issue #11775  that seems to be more similar to mine.

arnmar, 2021-11-20:
Removing __flash seems to fix the RGB issue for me on my DZ60RGB ANSI. Compiling with gcc 9.3.0 on WSL Ubuntu.

PR #15268 (2021-11-24): Remove use of __flash due to LTO issues

## Description

Due the interaction of `__flash` and LTO, it seems like the short term solution is to move to `PROGMEM` till the issue resolves.

## Types of Changes

- [x] Core
- [x] Bugfix
- [ ] New feature
- [ ] Enhancement/optimization
- [x] Keyboard (addition or update)
- [ ] Keymap/layout/userspace (addition or update)
- [x] Documentation

## Issues Fixed or Closed by This PR

* Fixes #14328

## Checklist

- [x] My code follows the code style of this project: [**C**](https://docs.qmk.fm/#/coding_conventions_c), [**Python**](https://docs.qmk.fm/#/c [... cut]
closed 2022-01-18 by moyi4681