qmk/qmk_firmware #14332: [Bug] Size to big?

Waples, 2021-09-06:
## Describe the Bug

I seem to be unable to compile my keymap (see my [fork](https://github.com/Waples/qmk_firmware/tree/crkbd-waples/keyboards/crkbd/keymaps/waples) ), because it's to big? Any idea how I can fix this, or is this a known bug?

```bash
$  qmk compile
Ψ Compiling keymap with make --jobs=1 crkbd:waples

QMK Firmware 0.14.9
Making crkbd/rev1 with keymap waples

...

Copying crkbd_rev1_waples.hex to qmk_firmware folder                                                [OK]
Checking file size of crkbd_rev1_waples.hex
 * The firmware is too large! 31030/28672 (2358 bytes over)
 [ERRORS]

make[1]: *** [tmk_core/rules.mk:473: check-size] Error 1
Make finished with errors

```

## System Information

 - Keyboard:  crkbd
   - Revision (if applicable): rev1 
 - Operating system: Linux
 - AVR GCC version: avr-gcc (GCC) 11.2.0

 - ARM GCC version: arm-none-eabi-gcc (Arch Repository) 11.2.0

 - QMK Firmware version: 0.14.9

 - Any keyboard related software installed? 
  no

## Additional Context

I seem to be unable to compile my keymap (see my [fork](https://github.com/Waples/qmk_firmware/tree/crkbd-waples/keyboards/crkbd/keymaps/waples) ), because it's to big? Any idea how I can fix this, or is this a known bug?

hobby-grade, 2021-09-07:
Is the firmware not simply too big for the available storage on the microcontroller? Or are you saying that it should fit but it still gives you the error?

Cipulot, 2021-09-07:
I'd advice to add `LTO_ENABLE` in your `rules.mk`.
The mentioned error, `* The firmware is too large! 31030/28672 (2358 bytes over)` is simly because the whole compiled code is bigger that the space available on the MCU.
You can remove some functions like RGB, which uses a lot of memory. If you don't want to give up RGB effects you might consider enabling only the effects that you actually are gonna use.

drashna, 2021-09-08:
Also, don't use sprintf/snprintf for oleds, if those are enabled, as that takes up a good chunk of firmware space. 

And rolling back to 8.4.0 for avr-gcc will help will the file size, too.  But I know that's a PITA for arch.

Waples, 2021-09-15:
> I'd advice to add `LTO_ENABLE = yes` in your `rules.mk`.
> The mentioned error, `* The firmware is too large! 31030/28672 (2358 bytes over)` is simply because the whole compiled code is bigger that the space available on the MCU.
> You can remove some functions like RGB, which uses a lot of memory. If you don't want to give up RGB effects you might want to consider enabling only the effects that you actually are gonna use.

How can you enable/disable them RGB effects? I don't need wave effects for example

Cipulot, 2021-09-15:
You can take a look [here](https://beta.docs.qmk.fm/using-qmk/hardware-features/lighting/feature_rgblight#example-usage-to-reduce-memory-footprint) on the QMK Docs. Basically using `#define RGBLIGHT_ANIMATIONS` compiles all the animations code into your final binary. Following the instructions in the docs you can look up the animations that you want to use and define only those. That way no superfluous code is compiled, taking up space in your firmware.
Also wpm drawing on oleds via `sprintf/snprintf` takes up some pretious space as drashna said before. So if you don't need/want it that's basically free space for you.

Waples, 2021-11-18:
I'm closing this issue out of **CRUFT**.

Will pick this up again when I've build the next keeb.

closed 2021-11-18 by Waples