qmk/qmk_firmware #14359: qmk固件能用最少的io实现更多的键吗 Can qmk firmware achieve more keys with the least io

jeromeli002, 2021-09-09:
qmk固件能用最少的io实现更多的键吗
如下地址 N个io脚可以实现N*(N-1)按键，lotbk蓝牙已经可用了
#define LESS_IO  //键盘阵列采用更少的IO口，不知道qmk是否支持，如哦支持应该怎么启用

Can qmk firmware achieve more keys with the least io
The following address N io pins can realize N*(N-1) button, lotbk bluetooth is already available
#define LESS_IO //The keyboard array uses fewer IO ports. I don’t know if qmk supports it. If so, how should the support be enabled?

https://baijiahao.baidu.com/s?id=1621173179116478690&wfr=spider&for=pc
https://oshwhub.com/yANgZEN/tiao-zhan-zui-di-cheng-ben-de-61

elfmimi, 2021-09-11:
QMK does not directly support that technique. however it is relatively easy to implement such tweak that is unique to individual keyboards.

```CUSTOM_MATRIX = lite``` is the key.

refer to [jones](https://github.com/qmk/qmk_firmware/tree/master/keyboards/jones) keyboard, which is built on the same idea.

https://github.com/qmk/qmk_firmware/blob/35bff470f73e5041f4e969f5dae5f60d4c1ea3e0/keyboards/jones/v03_1/rules.mk#L26-L27
https://github.com/qmk/qmk_firmware/blob/35bff470f73e5041f4e969f5dae5f60d4c1ea3e0/keyboards/jones/v03_1/config.h#L45-L46
https://github.com/qmk/qmk_firmware/blob/35bff470f73e5041f4e969f5dae5f60d4c1ea3e0/keyboards/jones/v03_1/v03_1.h#L22-L33
and [matrix.c](https://github.com/qmk/qmk_firmware/blob/35bff470f73e5041f4e969f5dae5f60d4c1ea3e0/keyboards/jones/v03_1/matrix.c)

myst729, 2021-10-18:
Wow~ Charlieplex, cool.

jeromeli002, 2021-10-18:
> 你用过二极管吗？
全部都有用二极管的，还是会这样 部分组合键有问题会冲突
All of them use diodes, but they will still be like this. Some key combinations will conflict with problems
[image]

myst729, 2021-10-19:
In regular (ortho) matrix, `#define DIODE_DIRECTION COL2ROW // or ROW2COL` acts like a virtual diode indeed. For round-robin matrix, each pin can be column or row in different key switches. So you need extra physical diodes to cover that part. Take a look at Jones' PCB design.

jeromeli002, 2022-06-02:
Is this matrix not usable after the update now? 
现在更新后这个矩阵不能用了吗

closed 2022-10-01 by github-actions[bot]