qmk/qmk_firmware #14377: How should I use the Key Overrides feature?

shelaf, 2021-09-10:
I want to create a keymap that allows to input the same character as a keycap engraving on the US layout keyboard while keeping the OS keyboard layout setting set to the Japanese keyboard. I've already achieved this by implementing [custom code](https://github.com/shelaf/qmk_firmware/blob/fix/keyboards/hhkb/ansi/keymaps/shela/action_pseudo.c), but the problem is that some key repeats are broken. Therefore, I would like to achieve this using the Key Overrides feature.

JP Keyboard:
[image]

US Keyboard:
[image]

Some examples I want to override:
|pressed|input with JP keyboard|want to input|replace to|
|-------|----------------------|-------------|----------|
|shift+2|"|@|[|
|shift+0|(Nothing)|)|shift+9|
|shift+[|`|{|shift+]|
|shift+'|*|"|shift+2|
|[|@|[|]|
|'|:|'|shift+7|

Code I wrote:
```
const key_override_t at_key_override = ko_make_basic(MOD_MASK_SHIFT, KC_2, KC_LBRC);
const key_override_t rprn_key_override = ko_make_basic(MOD_MASK_SHIFT, KC_0, S(KC_9));
const key_override_t lcbr_key_override = ko_make_basic(MOD_MASK_SHIFT, KC_LBRC, S(KC_RBRC));
const key_override_t dquo_key_override = ko_make_basic(MOD_MASK_SHIFT, KC_QUOT, S(KC_2));
const key_override_t lbrc_key_override = ko_make_with_layers_and_negmods(0, KC_LBRC, KC_RBRC, ~0, (uint8_t) ~MOD_MASK_SHIFT);
const key_override_t quot_key_override = ko_make_with_layers_and_negmods(0, KC_QUOT, S(KC_7), ~0, (uint8_t) ~MOD_MASK_SHIFT);

const key_override_t **key_overrides = (const key_override_t *[]){
    &at_key_override,
    &rprn_key_override,
    &lcbr_key_override,
    &dquo_key_override,
    &lbrc_key_override,
    &quot_key_override,
    NULL
}
```

This works fine in most cases. However, some corner cases do not work as intended.

|pressed|input|expected|
|-------|-----|--------|
|ctrl+'|ctrl+:|ctrl+shift+7|
|alt+'|alt+:|alt+shift+7|
|gui+'|gui+:|gui+shift+7|
|ctrl+[|ctrl+@|ctrl+]|
|alt+[|alt+@|alt+]|
|gui+[|gui+@|gui+]|

Same for modifier key combinations other than shift.

I need to remove at least [this line](https://github.com/qmk/qmk_firmware/blob/7277f1febe0e86e8a32beb779aa24e02bfdb6826/quantum/process_keycode/process_key_override.c#L119) to fix this issue, but I'm not sure if this is the intended behavior.
Please let me know if anyone has a good idea.  (tag: @JonasGessner)

JonasGessner, 2021-09-10:
If I understand correctly it seems like you need to change the following lines:

```
const key_override_t lbrc_key_override = ko_make_with_layers_and_negmods(0, KC_LBRC, KC_RBRC, ~0, (uint8_t) ~MOD_MASK_SHIFT);
const key_override_t quot_key_override = ko_make_with_layers_and_negmods(0, KC_QUOT, S(KC_7), ~0, (uint8_t) ~MOD_MASK_SHIFT);
```

The negative mod mask you set there contains all mods except shift. You'd want the opposite: Only shift should be a negative mod (negative mod = if that mod is down the override won't active. More info in the docs). So try getting rid of the bitwise NOT:

```
const key_override_t lbrc_key_override = ko_make_with_layers_and_negmods(0, KC_LBRC, KC_RBRC, ~0, MOD_MASK_SHIFT);
const key_override_t quot_key_override = ko_make_with_layers_and_negmods(0, KC_QUOT, S(KC_7), ~0, MOD_MASK_SHIFT);
```

shelaf, 2021-09-11:
Oh, you're right! After fixing them, everything worked as I intended.
The Key Overrides feature is very nice ! Thanks for your work !

closed 2021-09-11 by shelaf