qmk/qmk_firmware #14450: [Bug] ymdk_np21 RGB backlight issues

AlexX333BY, 2021-09-15:
I have issues with RGB backlight - it is red all the time. `_BL` layer of default keymap is not working to change backlight with default keys.
I've tried to fill every key of `_BL` keymap with alpha keys - they are working, so keys are working on this level.
I've tried to change animations in config.h, it didn't change anything (backlight is still red).
Backlight was working on default firmware (some rainbow effect was enabled, and I could toggle it with `Fn + /`).

zvecr, 2021-09-15:
The `BL_*` keycodes do not control RGB, and you should use <https://docs.qmk.fm/#/feature_rgblight?id=keycodes> instead.

closed 2021-09-15 by zvecr

AlexX333BY, 2021-09-16:
Hi @zvecr 
That worked, thanks. But shouldn't `default` keymap (as well as others included in repository) work without modification? I can update keymaps with `RGB_*` keycodes if this issue can be reopened