qmk/qmk_firmware #14458: [Bug] Compatibility with split keyboards ATMega32U4-AU

nicolkill, 2021-09-16:
# atmega32u4-au doesnt works with split keyboards

## Describe the Bug
The Pro Micro with the atmega32u4-au doesnt works with split keyboards, specifically crkbd
the main part works but the connected by trrs dont signs any response

## System Information

 - Keyboard:  
   - Revision (if applicable): crkbd v2 and v3
 - Operating system: Linux and windows tested
 - AVR GCC version: 5.4.0 

 - ARM GCC version:(15:6.3.1+svn253039-1build1) 6.3.1 20170620

 - QMK Firmware version: 0.14.16

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

drashna, 2021-09-16:
It probably needs `#define SPLIT_USB_DETECT` as VBUS detection isn't supported by the clones in question.

Daveyr, 2022-06-16:
I'm having similar issues with pro micro clones in my modified Lily58. The master side works fine but the off-hand side seizes up eventually and requires a reset to work. A minute later, it hangs again. `EE_HANDS` tested, also `SPLIT_USB_DETECT` but no help...

keyboard-magpie, 2022-08-03:
Last comment is not related to the initial issue and should be discussed elsewhere.

closed 2022-08-03 by keyboard-magpie