qmk/qmk_firmware #14467: [Bug] unicode map stopped working

alinelena, 2021-09-16:
Unicode map stopped working

## Describe the Bug

I am using unicode map to enter some math characters ∑ and similiar.
It worked fine up to recent 0.14 update.

I git bisect it and it stopped working at this commit

```
git bisect good                                                                                                                       15/09/21 06:44:02
e756a21636149ad47c19c659d04be93cf3071dab is the first bad commit
commit e756a21636149ad47c19c659d04be93cf3071dab
Author: Donald Kjer <don.kjer@gmail.com>
Date:   Mon Aug 23 15:15:34 2021 -0700
 
    eeprom_stm32: implement high density wear leveling (#12567)
```

as a workaround I noticed than when I have this
  #define UNICODE_SELECTED_MODES UC_LNX, UC_MAC, UC_WINC                                                                                                                        
  is not working but
#define UNICODE_SELECTED_MODES UC_LNX 
it is.

all this was tested on stm32f103 and gd32f303 (both are compatible)

LucasEngel2, 2022-02-09:
Having a very similar issue, tring to type ä, ö, ü, but I'm just getting the unicodes printed out.
00e4 00f6 00fc.

alinelena, 2022-02-09:
if you are on linux, just restart the terminal

closed 2022-07-13 by github-actions[bot]