qmk/qmk_firmware #14509: [Bug] `qmk info` throws an error if no `keymap` is provided / defined

j4ckofalltrades, 2021-09-19:
## Describe the Bug

Running `qmk info -kb <keyboard>` throws an error if there is no `-km | --keymap` argument provided or no default keymap is configured via `qmk config`.

```sh
$ qmk info -kb arisu
Keyboard Name: Arisu
Manufacturer: Fate
Website: https://github.com/FateNozomi/arisu-pcb
Maintainer: FateNozomi
Keyboard Folder: arisu
Layouts: LAYOUT
Processor: atmega32u4
Bootloader: atmel-dfu
<class 'TypeError'>
☒ unsupported operand type(s) for /: 'PosixPath' and 'NoneType'
Traceback (most recent call last):
  File "/home/jordan/.local/lib/python3.7/site-packages/milc/milc.py", line 523, in __call__
    return self.__call__()
  File "/home/jordan/.local/lib/python3.7/site-packages/milc/milc.py", line 528, in __call__
    return self._subcommand(self)
  File "/home/jordan/projects/personal/qmk_firmware/lib/python/qmk/decorators.py", line 27, in wrapper
    return func(*args, **kwargs)
  File "/home/jordan/projects/personal/qmk_firmware/lib/python/qmk/decorators.py", line 47, in wrapper
    return func(*args, **kwargs)
  File "/home/jordan/projects/personal/qmk_firmware/lib/python/qmk/cli/info.py", line 188, in info
    show_keymap(kb_info_json, title_caps)
  File "/home/jordan/projects/personal/qmk_firmware/lib/python/qmk/cli/info.py", line 24, in show_keymap
    keymap_path = locate_keymap(cli.config.info.keyboard, cli.config.info.keymap)
  File "/home/jordan/projects/personal/qmk_firmware/lib/python/qmk/keymap.py", line 287, in locate_keymap
    if (keymap_dir / keymap / 'keymap.c').exists():
TypeError: unsupported operand type(s) for /: 'PosixPath' and 'NoneType'
```

I tried printing out the keymap argument by adding `print('keymap (-km | --keymap): ' + cli.config_source.info.keymap)` and the result was `keymap (-km | --keymap): argument`.

## System Information

 - Keyboard:  
   - Revision (if applicable):  
 - Operating system:
 - AVR GCC version: 

 - ARM GCC version: 

 - QMK Firmware version:

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

fauxpark, 2021-09-19:
I would say this should be filed in https://github.com/qmk/qmk_cli, except that I don't get this error with MILC 1.5.0 - only after updating to the latest version (1.6.4) do I get the same thing, and it also occurs in MSYS with:
```
TypeError: unsupported operand type(s) for /: 'WindowsPath' and 'NoneType'
```
So it is either a MILC bug, or some interaction between MILC and the QMK CLI.

j4ckofalltrades, 2021-09-19:
Thanks for looking into it, I've filed a new issue in [qmk_cli](https://github.com/qmk/qmk_cli/issues/98) and referenced this one. Marking as closed.

closed 2021-09-19 by j4ckofalltrades