qmk/qmk_firmware #14549: Couldn't install QMK on Ubuntu 20.04

mianqi2016, 2021-09-22:
I want to install QMK on Ubuntu 20.04 as it was told here - https://docs.qmk.fm/#/newbs_getting_started
My questions couldn't be sent by other methods, so I sent them here:

1. After "python3 -m pip install --user qmk", must I select one of the "Community Packages"?
2. I tried "sudo pacman -S qmk" and "yay -S qmk-git", they were all prompted:"command not found", how to fix this?

greyhatmiddleman, 2021-09-22:
> 2\. I tried "sudo pacman -S qmk" and "yay -S qmk-git", they were all prompted:"command not found", how to fix this?

This is for Arch Linux. You don't needs these commands.

If you were able to running the python command you should be able to run `qmk setup`, step 3 of the guide and continue from there.

mianqi2016, 2021-09-24:
I have ran "qmk setup", here was the newest prompt:
"Ψ Git branch: master
⚠ Git has unstashed/uncommitted changes.
⚠ The official repository does not seem to be configured as git remote "upstream"."
and:
"Ψ All dependencies are installed.
Ψ Found arm-none-eabi-gcc version 9.2.1
Ψ Found avr-gcc version 5.4.0
Ψ Found avrdude version 6.3-20171130
Ψ Found dfu-util version 0.9
Ψ Found dfu-programmer version 0.6.1
Ψ Submodules are up to date.
Ψ QMK is ready to go, but minor problems were found
"
How to go on?

greyhatmiddleman, 2021-09-24:
Which keyboard are you trying to compile a firmware to?

Once you find the keyboard name, under the `keyboards` folder, you should be able to run:
`qmk compile -kb <keyboard> -km default`

For example the keyboard that I have is a crkbd, so for me to test I would run
`qmk compile -kb crkbd -km default`

At that point if you are able to compile a firmware, you should read through the entire tutorial and FAQ to understand how to customize your keyboard. 

Have fun and good luck.

mianqi2016, 2021-09-25:
I saw your keyboard here - https://github.com/foostan/crkbd, nice!
But, I mean needn't I fix such prompts?
warning Git has unstashed/uncommitted changes.
warning The official repository does not seem to be configured as git remote "upstream"."

greyhatmiddleman, 2021-09-25:
If you are not planning to push changes to the general repository, these warnings don't impact you.

mianqi2016, 2021-10-04:
Now, I went into flashing. It was said here - https://beta.docs.qmk.fm/using-qmk/guides/flashing/flashing#stm32duino:
"
To ensure compatibility with the STM32-DFU bootloader, make sure this block is present in your rules.mk (optionally with apm32-dfu instead):
`# Bootloader selection
BOOTLOADER = stm32-dfu`
"
When I opened the file "rules.mk" under "/.../crkbd", the first line was:
`# MCU name
MCU = atmega32u4

# Bootloader selection
BOOTLOADER = caterina
`
Should I changed then into:
`# MCU name
MCU = stm32g474re

# Bootloader selection
BOOTLOADER = stm32-dfu
`?
Since I used a "NUCLOE-G474RE" for experiment.

greyhatmiddleman, 2021-10-04:
Not familiar with the NUCLOE-G474RE, but first I recommend verifying it is one of the [compatible controllers](https://beta.docs.qmk.fm/developing-qmk/c-development/compatible_microcontrollers). It doesn't seem stm32g474re is a supported controller but I could be wrong. 

If you setup your qmk accordingly you should able to run the following:
```qmk new-keymap -kb crkbd```

It should create a folder in the ```crkbd\keymaps\{user.keymap}```
> user.keymap based on your qmk setup.

There you can change/modify existing variables based on the controller under the ``crkbd\keymaps\{user.keymap}\rules.mk`` file. Its better to change settings based on the user keymap level than at the keyboard level. For your controller (if supported), should contain:
```
MCU = stm32g474re
BOOTLOADER = stm32-dfu
```

Here more on the [keyboard guidelines](https://beta.docs.qmk.fm/developing-qmk/c-development/hardware_keyboard_guidelines) to help you with your project.

mianqi2016, 2021-10-05:
I found STM32G474 was supported here - [Compatible Microcontrollers](https://beta.docs.qmk.fm/developing-qmk/c-development/compatible_microcontrollers), and:
1. created a folder in the crkbd\keymaps\{user.keymap}
2. added "
MCU = stm32g474re
BOOTLOADER = stm32-dfu
" into "crkbd\keymaps\{user.keymap}\rules.mk"

After typed:"qmk flash -kb crkbd -km crkbd_MQ", it was said:
"
QMK Firmware 0.14.16
Making crkbd/rev1 with keymap crkbd_MQ and target flash

build_keyboard.mk:404: tmk_core/protocol/avr.mk: No such file or directory
make[1]: *** No rule to make target 'tmk_core/protocol/avr.mk'.  Stop.
Make finished with errors
make: *** [Makefile:539: crkbd/rev1:crkbd_MQ:flash] Error 1
"

why it made target for "...avr.mk",?

mianqi2016, 2021-10-05:
After compiling, I found file “crkbd_rev1_default.hex” was created under directory “qmk_firmware”. After typed “flash...”, file “crkbd_rev1_crkbd_MQ.hex” was created.

T got hint here - [make Targets](https://beta.docs.qmk.fm/using-qmk/guides/flashing/flashing#make-targets-3), since I have ST-Link dongle, so I just programmed NUCLEO G474RE with file "crkbd_rev1_crkbd_MQ.hex" by STM32CubeProg, it was said succeed, so as I understood, now the NUCLEO G474RE is a crkbd controller, is that true?

mianqi2016, 2021-10-05:
I pulled out and plug in the board again, while after “lsbusb”, it was always shown:
Bus 003 Device 006: ID 0483:374e STMicroelectronics STLINK-V3
and, I could see the led was still blinking – I programmed “Blink” before QMK, so if it was a QMK firmware on it now, why didn’t it be recognized as a  “STLINK-V3” not “crkbd”? Even it was because the NUCLEO board has a “STLINK-V3” as interface between itself and pc, why the “Blink” code was still running?

greyhatmiddleman, 2021-10-05:
> I pulled out and plug in the board again, while after “lsbusb”, it was always shown: Bus 003 Device 006: ID 0483:374e STMicroelectronics STLINK-V3 and, I could see the led was still blinking – I [...]

It doesn't seem that the controller is flashed properly. 

I recommend ```qmk clean``` then the following. 
```
qmk compile -kb <keyboard> -kb <keymap>
# if you get no errors, then run
qmk flash -kb <keyboard> -kb <keymap>
```

Totally not familiar with that specific ARM controller, and depending on the pinouts, you might need to create a totally new custom keyboard because from what I understand the crkbd is specific to pro micro (or pro micro like) pinouts. That being said, the standard crkbd keyboard configuration would not be sufficient for your ARM controller. 

Looking at [this](https://www.mouser.tw/ProductDetail/STMicroelectronics/NUCLEO-8S207K8?qs=Cb2nCFKsA8oALqAmKKuFwA%3D%3D&mgh=1&gclid=CjwKCAjwzOqKBhAWEiwArQGwaDoCgCeptkQUITgbNtY2IoGUAJ5YEaU0jtOt4PC9FGjgLFgZdBZ8BxoCOQIQAvD_BwE), assuming that is the same MCU, it seems that the pinouts might be different. 

Even thought this is referenced to AVR controller, you might have to do something similar with your ARM controller [AVR Processors](https://beta.docs.qmk.fm/developing-qmk/c-development/hardware_avr). You'll need to setup the Hardware Config, Keyboard Matrix pinouts etc. 

Not sure if you were able to test your keyboard after you built it, but I highly suggest getting 2 Pro Micros and compile/flash the default keymaps to make sure your keyboard works. After that you might have to do a lot of customized development to get NUCLEO G474RE working.

greyhatmiddleman, 2021-10-07:
Cool to see that you got your answers. You might want to consider closing this issue since it seems that the QMK install is working fine at this point. I recommend open another case in regards to the controller. There might be others who might be more experienced in that area.

mianqi2016, 2021-10-11:
OK, thank you for your advice and help.

closed 2021-10-11 by mianqi2016