qmk/qmk_firmware #14559: Adding RGB blacklight support to Ducky One2 mini

dfireBird, 2021-09-22:
The PR #11112, which added support to Ducky One2 mini, had support for RGB blacklight initially and then  removed later.

This issue is track to what's blocking it for adding that support?

CraigA-ZA, 2021-12-27:
What's blocking it is that nobody is actively working on it. The initial implementation was written years ago by GitWellBack and they aren't actively working on the project anymore, from what I can tell. Their implementation isn't compatible with the current ChibiOS version. Someone else will have to take on the task. Fishman was the other guy who adopted the project, but I don't believe he is working on getting the RGB stuff working as it required a rewrite of GitWellBack's stuff, from what he said in a comment somewhere.

jpfelgueiras, 2022-04-26:
Is any one working on this? 
How could help and what skill are needed?

paintchip, 2022-07-05:
It looks like the driver was written and a PR was created, but it wasn't merged because the implementation wasn't platform agnostic enough. According to the conversation, the driver has to support both ARM and AVM or live on its own in the keyboard directory (I'm not sure how it's applied in that case).

Fishman's last comment was that he moved the driver to the keyboard directory, but there is no corresponding commit and he closed the PR shortly after.

https://github.com/qmk/qmk_firmware/pull/11022

f7urry, 2022-07-05:
i was try @fishman qmk forked with branch [duckybuild](https://github.com/fishman/qmk_firmware/tree/duckybuild) on my ducky one2sf but the driver seems need some improvement because i got difficult to lightup 4th row LED [Here](https://drive.google.com/file/d/1R2pZosn9NgjJn-0TlXWrvic3uO7mZdmk/view?usp=sharing), 1-3rd row led just working fine and the matrix just reversed, still need time to figure out

CraigA-ZA, 2022-07-05:
@Khairullah I had the exact same experience as you when using the LED driver he wrote. I changed the pin matrix same as you did, but also couldn't get the bottom row to work

tzarc, 2022-10-05:
Stale, no indication of anyone working on it. Closing.

closed 2022-10-05 by tzarc