qmk/qmk_firmware #14619: [Bug] KP_END sometimes doesn't work on first hits (YMDK Wings Hotswap)

tedbarth, 2021-09-27:
## Describe the Bug

After resuming from suspension (sometimes) or right after starting of the PC (always) the `KP_END` switch on my *Wings Hotswap* does not work with the following keymap. Multiple key presses do not get recognized then. After some runtime (literally while writing this text) the behavior appears again.

To make the button work again sometimes it is enough to long press the key (after two seconds the cursor jumps) or switching layers with MO(1) bound to another key. The key binding `KC_PGDN` on the same switch but other layer always works.

Maybe interesting is that on the default keymap that was active before the same switch was bound to MO(1).

_Could this be a problem with EEPROM? Can I reset it?_
_Could this be a hardware problem?_

## System Information

 - Keyboard:  YMDK Wings Hotswap
 - Operating system: Windows 10 + Ubuntu 21.04
 - AVR GCC version:  5.4.0
 - ARM GCC version: 9.2.1 20191025
 - QMK Firmware version: 9.2.1 20191025
 - Any keyboard related software installed? No.

## Changed files against repo version of default keymap for this keyboard (+ no VIA enabled). I copied the default keymap and as a different one and changed the bindings. 

keymap.c
```
/* Copyright 2021 alittlepeace
...
*/

#include QMK_KEYBOARD_H

const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {
  [0] = LAYOUT_all(
    KC_ESC,  KC_1,    KC_2,    KC_3,    KC_4,    KC_5,    KC_6,    KC_7,    KC_8,    KC_9,    KC_0,    KC_MINS, KC_EQL,  KC_BSPC,                KC_DEL,  
    KC_TAB,  KC_Q,    KC_W,    KC_E,    KC_R,    KC_T,    KC_Y,    KC_U,    KC_I,    KC_O,    KC_P,    KC_LBRC, KC_RBRC, KC_BSLS,                KC_HOME,
    MO(1),   KC_A,    KC_S,    KC_D,    KC_F,    KC_G,    KC_H,    KC_J,    KC_K,    KC_L,    KC_SCLN, KC_QUOT, KC_ENT,                          KC_END,
    KC_LSFT, KC_Z,    KC_X,    KC_C,    KC_V,    KC_B,    KC_N,    KC_M,    KC_COMM, KC_DOT,  KC_SLSH, KC_RSFT,                    KC_UP,
    KC_LCTL,          KC_LALT,          KC_SPC,  KC_LGUI,          KC_SPC,           KC_RALT,                            KC_LEFT,  KC_DOWN, KC_RGHT 
  ),

  [1] = LAYOUT_all(
    KC_GRAVE, KC_F1,    KC_F2,   KC_F3,   KC_F4,   KC_F5,   KC_F6,   KC_F7,   KC_F8,   KC_F9,   KC_F10,  KC_F11,  KC_F12,  _______,              KC_INS,
    _______,  RGB_MOD,  _______, RGB_TOG, _______, _______, _______, _______, _______, _______, KC_PSCR, _______, _______, _______,              KC_PGUP,
    _______,  RGB_SAI,  RGB_VAI, RGB_SPI, RESET,   _______, _______, _______, _______, _______, _______, _______, KC_SLEP,                       KC_PGDN,
    _______,  _______,  _______, _______, _______, _______, _______, _______, _______, _______, _______, _______,                   KC__VOLUP, 
    _______,            _______,          _______, _______,          KC_MPLY,          _______,                            KC_MPRV, KC__VOLDOWN, KC_MNXT 
  )    
};
```

@alittlepeace You might be interested in this.

tedbarth, 2021-09-28:
Apparently this problem occurs only on my Desktop PC, not on my working Laptop with Ubuntu 21.04 (can't test Windows there).

haleyatyler, 2021-11-14:
Hi there. I just finished building my YMDK Wings (hotswap) today. I didn't know if I should make a new issue post or not, but I am having troubles with the keymap JSON file in VIA. When initially plugging the keyboard in, all the keys work perfectly fine and the RGB is static red.

However, when attempting to import the keymap in VIA, it does not work. It continues to say "searching for devices".
I have attempted to re-flash and create a new keymap on QMK Config, but they do not work either. So my keyboard is stuck with static red RGB and I am unable to/ customize my keyboard layers / macros through VIA. Is there any way someone can reach out to me on Discord about this? I have exhausted all of my resources with no success. My Discord username is haleychu#0385. Any help would be greatly appreciated. Thanks!

@alittlepeace

closed 2021-12-24 by drashna