qmk/qmk_firmware #14621: [Bug] Implicit declaration of `mousekey_console()` after pulling master

mcp292, 2021-09-27:
## Describe the bug
After pulling from master I get an `implicit declaration of function` error:
```text
Compiling: quantum/command.c                                                                       quantum/command.c: In function 'command_console':
quantum/command.c:561:13: error: implicit declaration of function 'mousekey_console' [-Werror=implicit-function-declaration]
  561 |             mousekey_console(KC_SLASH /* ? */);
      |             ^~~~~~~~~~~~~~~~
quantum/command.c: At top level:
quantum/command.c:629:6: error: conflicting types for 'mousekey_console'; have '_Bool(uint8_t)' {aka '_Bool(unsigned char)'}
  629 | bool mousekey_console(uint8_t code) {
      |      ^~~~~~~~~~~~~~~~
quantum/command.c:561:13: note: previous implicit declaration of 'mousekey_console' with type 'int()'
  561 |             mousekey_console(KC_SLASH /* ? */);
      |             ^~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
 [ERRORS]
 |
 |
 |
make[1]: *** [tmk_core/rules.mk:443: .build/obj_crkbd_rev1_mcp292/quantum/command.o] Error 1
Make finished with errors
make: *** [Makefile:539: crkbd/rev1:mcp292:flash] Error 1
```

Git blame links it to [this commit/PR](https://github.com/qmk/qmk_firmware/pull/11842), but I don't find it in my local logs nor in the [github commits page for master](https://github.com/qmk/qmk_firmware/commits/master). This makes sense because the PR was merged into develop. So I now search the master logs for "develop" to find the merge, and find nothing.

If found I would revert the commit and test to verify it is the cause of error.

## System Information

 - Keyboard:  crkbd
 - Operating system: Arch Linux 
 - AVR GCC version: 11.2.0
 - ARM GCC version: 11.2.0
 - QMK Firmware version:  0.14.16

mcp292, 2021-09-27:
Everything works fine when reverting to c2f227d2a8 (my last pulled commit before today).

sigprof, 2021-09-27:
For some reason `quantum/command.c` has this:
```c
#if defined(MOUSEKEY_ENABLE) && !defined(MK_3_SPEED)
#    include "mousekey.h"
#endif
```
and this:
```c
#if defined(MOUSEKEY_ENABLE) && !defined(MK_3_SPEED)
static bool mousekey_console(uint8_t code);
#endif
```
I guess you have `MK_3_SPEED` defined? Apparently the “`mousekey_console()` is now enabled regardless of `MK_3_SPEED`” part of #11842 was not 100% complete.

And the commit 383fae55c5d53626eda210298f15fa6810510d5e is actually in the history between c2f227d2a87d6258bca6d2f96b6fc475e2fe54f5 and current master, so looks like git blame pointed to it correctly.

mcp292, 2021-09-27:
Yes I do have `MK_3_SPEED` defined. Oh good catch! I didn't see the `#include`!

mcp292, 2021-09-27:
Odd I still don't find the commit in history even when searching by hash in my local logs.