qmk/qmk_firmware #14628: [Bug]  Unable to compile production or bootloader target - gcc make error

AABatteries, 2021-09-28:
## Describe the Bug

I am trying to compile a production target for keyboard 9key (using freshly forked repo, available on my github - just minor changes, swapped COL and ROW pins, also bootloader change to QMK DFU) and while I'm able to compile the standard 9key fw with default layout using `make 9key:default` (and it works great), I am getting errors while doing these commands:
`make 9key:default:production` or `make 9key:default:bootloader`

I'm attaching a snippet of the output that I think is relevant below:
`
[LNK]     : Linking object files into "BootloaderDFU.elf"
avr-gcc obj/BootloaderDFU.o obj/Descriptors.o obj/BootloaderAPI.o obj/BootloaderAPITable.o obj/HIDParser.o obj/Device_AVR8.o obj/EndpointStream_AVR8.o obj/Endpoint_AVR8.o obj/Host_AVR8.o obj/PipeStream_AVR8.o obj/Pipe_AVR8.o obj/USBController_AVR8.o obj/USBInterrupt_AVR8.o obj/ConfigDescriptors.o obj/DeviceStandardReq.o obj/Events.o obj/HostStandardReq.o obj/USBTask.o -o BootloaderDFU.elf -lm -Wl,-Map=BootloaderDFU.map,--cref -Wl,--gc-sections -Wl,--relax -mmcu=atmega32u4 -flto -fuse-linker-plugin -Wl,--section-start=.text=0x7000 -Wl,--section-start=.apitable_trampolines=0x7FA0 -Wl,--undefined=BootloaderAPI_Trampolines -Wl,--section-start=.apitable_jumptable=0x7FE0 -Wl,--undefined=BootloaderAPI_JumpTable -Wl,--section-start=.apitable_signatures=0x7FF8 -Wl,--undefined=BootloaderAPI_Signatures
/usr/lib/gcc/avr/5.4.0/../../../avr/bin/ld: section .apitable_trampolines loaded at [0000000000007fa0,0000000000007fb7] overlaps section .data loaded at [0000000000007f2c,0000000000007fa9]
collect2: error: ld returned 1 exit status
make[2]: *** [../../LUFA/Build/DMBS/DMBS/gcc.mk:251: BootloaderDFU.elf] Error 1
make[1]: *** [tmk_core/avr.mk:170: bootloader] Error 2
Make finished with errors
make: *** [Makefile:539: 9key:default:production] Error 1

`

## System Information

 - Keyboard:  DIY 9 key macro pad - Arduino Pro Micro
 - Operating system: pop!_OS 21.04
 - AVR GCC version: 5.4.0

 - ARM GCC version: 9.2.1

 - QMK Firmware version: 0.14.16

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

Any help would be greatly appreciated, thank you!

AABatteries, 2021-09-28:
I've searched the QMK repo for similar issues - could this be related to #6719 ?

fauxpark, 2021-09-28:
AVR code compiles somewhat larger after GCC 8.x for whatever reason. Only the GCC people would know why. You'll just have to downgrade avr-gcc unfortunately.

AABatteries, 2021-09-28:
> AVR code compiles somewhat larger after GCC 8.x for whatever reason. Only the GCC people would know why. You'll just have to downgrade avr-gcc unfortunately.

What would be the best way to go about this? Thank you!

tzarc, 2022-08-29:
You'd have to consult the pop!_OS documentation to figure out how to downgrade, I'm afraid.

closed 2022-08-29 by tzarc