qmk/qmk_firmware #14658: [Bug] Media keycode(and some others) doesn't work well with Oneshot layer

ForsakenRei, 2021-09-30:
OSL didn't work on some of the keycode avaiable but works for custom keycode.

## Describe the Bug
I use Fn as a Oneshot layer key with some media keys like `KC_MPLY`, `KC_VOLU` and some other keycode like `KC_CALC` and `KC_LOCK` on that layer. I also put a custom keycode `C(S(KC_ESC))` for test purpose.

If I tap Fn, the one shot layer was activated(LED indicator for that layer lights up) but `KC_MPLY` or `KC_CALC` won't work, while my custom keycode `C(S(ESC))` and `KC_LOCK` works fine. `KC_VOLU` worked once but acted as stuck to turn my volume all the way down to mute.

If I hold Fn, everything just works fine. And switch to that layer everything works fine, too.

The issue was found originally when I use a tap dance on my Fn to activate the one shot layer, after that I tried OSL(1) but it doesn't work either.

## System Information

 - Keyboard: GMMK Pro ANSI(STM32)
 - Operating system: Windows 10
 - AVR GCC version: 8.4.0

 - ARM GCC version: 10.1.0

 - QMK Firmware version: 0.13.23
 tested on both master and dev branch

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

sigprof, 2021-09-30:
This might be a combination of two problems:

1) The code which handles turning the one shot layer off fakes a key release event immediately after processing the key press event. Because those events are separate, the release event bypasses `TAP_CODE_DELAY` (which is the usual solution for the media keys not registering properly), and therefore can trigger the problem 2 below.
https://github.com/qmk/qmk_firmware/blob/52cb19713b794be130563ac20923e8e70e83971e/quantum/action.c#L728-L738
I'm not sure whether that piece of code is really needed (or maybe it should be inside `#ifdef STRICT_LAYER_RELEASE`) — usually the source layer cache ensures that the key release event would be handled using the same keycode that was used for the matching key press event, and therefore just turning off the one shot layer should be enough.
2. The ChibiOS implementation of the code which actually sends the events for media keys to the host may misbehave if it is invoked before the previous event has been read by the host. PR #14643 attempts to fix that, but it is still not decided which way to fix the problem would be the most correct one.

ForsakenRei, 2021-10-01:
@sigprof Thanks for the heads up. Actually @Dasky discussed his issue with me earlier in discord server, though the symptoms are different but it seems they are quite related. I will keep an eye on that pull request.

closed 2021-10-19 by ForsakenRei