qmk/qmk_firmware #14673: [Bug] Dactyl Manuform - non-master split unrecognized after flashing

dstanberry, 2021-10-02:
## Describe the Bug

Since #14344 (I think this is the commit that has introduced the discrepancy for me), flashing the dactyl_manuform_5x7 causes the non-master half to become unrecognizable. Even flashing the default keymap that is shipped results in the same outcome.
Both halves work as expected when plugged in (usb). This issue is not present when reverting to a previous issue and re-flashing.

## System Information

 - Keyboard:  
   - Dactyl Manuform 5x7
 - Operating system:
   - WSL
 - AVR GCC version: 
   - 8.5.0
 - ARM GCC version: 
   - 8.5.0
 - QMK Firmware version:
    - 0.14.9 (issue present)
    - 0.13.34 (working prior to merging master rev)
 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

sigprof, 2021-10-02:
There was a bug in PR #14344; however, PR #14352 was supposed to fix that bug (together with some bugs in other keyboards). Did you check the most recent QMK code?

sigprof, 2021-10-02:
Also the PR that you linked is for `handwired/dactyl`, not any variation of `handwired/dactyl_manuform` — which keyboard do you actually have? If you have the variation with a Pro Micro (or a compatible controller) in each half, then maybe the problem is that you just need to reflash both of them with the new firmware version — flashing only the master half does not work after the main QMK code has been significantly changed.

dstanberry, 2021-10-02:
I have a dactyl_manuform with a Pro Micro in each half. I haven't tried flashing both halves, I'll try that next to see if it resolves things. If it does, maybe the README for the manuform  should be updated to reflect this nuance for the Pro Micro.

dstanberry, 2021-10-02:
@sigprof I've flashed both the same image to both but this hasn't changed the behavior.

closed 2021-10-03 by dstanberry