qmk/qmk_firmware #14686: [Bug] Cannot control RGB Matrix manually, colors repeatedly overwritten with solid color under RGB_MATRIX_NONE mode

Myridium, 2021-10-04:
## Describe the Bug

The RGB matrix mode `RGB_MATRIX_NONE` interferes with manual RGB control. Unable to set RGB states manually.

Setting `rgb_matrix_mode(RGB_MATRIX_NONE);` results in a constant red colour under every key. When I set an LED color using `rgb_matrix_set_color`, the new color setting applies for an instant before being overwritten again by the red colour. This makes it impossible to control the LEDs manually.

 - Keyboard: ZSA Moonlander
 - QMK Firmware version: 0.14.16

Myridium, 2021-10-04:
Investingating a bit more, I have found that `rgb_matrix_mode(RGB_MATRIX_NONE)` actually sets the matrix mode to `RGB_MATRIX_SOLID_COLOR` instead. The output of `rgb_matrix_get_mode()` shows that the mode was not set to `0` but `1` instead.

drashna, 2021-10-04:
The "none" mode is special and is meant only to be used internally.  Using `rgb_matrix_mode` will ensure that it is not used, and that the solid color mode is used instead.

Myridium, 2021-10-04:
> The "none" mode is special and is meant only to be used internally. Using `rgb_matrix_mode` will ensure that it is not used, and that the solid color mode is used instead.

There's nothing I found in the QMK documentation to explain what the roles of the different matrix modes are, so naturally I assumed that `RGB_MATRIX_NONE` meant no lighting-- e.g. do not control the LEDs on the board automatically. Then I expected to be able to control them manually.

Perhaps this should be taken as an issue with the documentation clarity.

atomgomba, 2021-11-16:
Yes, the `RGB_MATRIX_NONE` mode is misleading and it seems it is not well documented. I'm also trying to disable automatic lighting while keeping manual control enabled and I tried this instead to no avail:

```c
void keyboard_post_init_user(void) {
    rgb_matrix_enable_noeeprom();
    rgb_matrix_mode_noeeprom(RGB_MATRIX_SOLID_COLOR);
    rgb_matrix_set_color_all(RGB_OFF);
}
```
There is still a red underglow right after boot.

JFT70, 2021-11-16:
Out of curiosity on which keyboard does this occur?

atomgomba, 2021-11-16:
@JFT70 This one has a SN32F248B (Sonix)

drashna, 2021-11-17:
@atomgomba are you using `rgb_matrix_indicators_user` for setting the leds individually, or?

atomgomba, 2021-11-17:
@drashna Yes, I set the color of some individual keys in `rgb_matrix_indicators_kb` and this was my motivation to disable the automatic lighting effect

drashna, 2023-10-12:
set the flags to "none".

merlink01, 2024-11-18:
Sorry what flag? Could you give a more detailed description?
I Tried all soutions above now in Lemokey P1 Pro and nothing helps.

EBADBEEF, 2025-05-01:
Here is how I solved it.
```
enum handle_rgb_action {
    ACTION_INIT,
    ACTION_TOGGLE,
    ACTION_ON,
    ACTION_OFF,
};
static bool handle_rgb(enum handle_rgb_action action, bool pressed)
{
    static rgb_config_t rgb_last_config;

    const led_flags_t flags_on = LED_FLAG_MODIFIER | LED_FLAG_KEYLIGHT;
    const led_flags_t flags_off = LED_FLAG_NONE;
    const uint8_t mode_off = RGB_MATRIX_SOLID_COLOR;

    if (!pressed) {
        return false;
    }

    if (action == ACTION_INIT) {
        rgb_last_config = rgb_matrix_config;
        if (rgb_matrix_get_flags() == flags_off)
            rgb_matrix_mode_noeeprom(mode_off);
        else
            rgb_matrix_set_flags_noeeprom(flags_on);
        rgb_matrix_enable_noeeprom();
        return false;
    }

    if (action == ACTION_TOGGLE) {
        if (rgb_matrix_get_flags() == flags_off)
            action = ACTION_ON;
        else
            action = ACTION_OFF;
    }

    if (action == ACTION_ON) {
        rgb_matrix_config = rgb_last_config;
        rgb_matrix_set_flags(flags_on);
    } else if (action == ACTION_OFF) {
        rgb_last_config = rgb_matrix_config;
        rgb_matrix_set_flags(flags_off);
        rgb_matrix_mode_noeeprom(mode_off);
    }

    return false;
}

void keyboard_post_init_kb(void)
{
    handle_rgb(ACTION_INIT, true);
}

bool process_record_kb(uint16_t keycode, keyrecord_t *record)
{
    if (process_record_user(keycode, record) == false)
        return false;

    if (keycode == QK_RGB_MATRIX_TOGGLE) {
        return handle_rgb(ACTION_TOGGLE, record->event.pressed);
    } else if (keycode == QK_RGB_MATRIX_ON) {
        return handle_rgb(ACTION_ON, record->event.pressed);
    } else if (keycode == QK_RGB_MATRIX_OFF) {
        return handle_rgb(ACTION_OFF, record->event.pressed);
    }

    return true;
}

bool rgb_matrix_indicators_advanced_kb(uint8_t led_min, uint8_t led_max)
{
    if (rgb_matrix_indicators_advanced_user(led_min, led_max) == false)
        return false;

    if (host_keyboard_led_state().caps_lock)
        rgb_matrix_set_color(RGB_INDEX_CAPS, RGB_RED);
    else
        rgb_matrix_set_color(RGB_INDEX_CAPS, RGB_OFF);

    if (keymap_config.no_gui)
        rgb_matrix_set_color(RGB_INDEX_WIN_LOCK, RGB_RED);
    else
        rgb_matrix_set_color(RGB_INDEX_WIN_LOCK, RGB_OFF);

    return true;
}
```

I don't want the indicators to be used in the RGB effect (I want them to be indicators!).

Instead of disabling rgb matrix (which would disable the indicators), set the mode to solid color and flags to LED_FLAG_NONE. The solid color mode is not saved to eeprom, but the flags are. No LEDs will be turned on by an animation effect (assuming they call LED_MATRIX_TEST_LED_FLAGS). On boot, detect what state we are in by checking flags in eeprom. The code is careful to never overwrite the actual effect/color settings in eeprom even when the lighting mode is disabled.