qmk/qmk_firmware #14700: "Modifier with Toggle Swith" behavior, it's posible?... how?

senseimitz, 2021-10-04:
Hi Jack Humbert, first of all congratulations on your work and thank you for sharing your knowledge.

3 months ago I met a "Planck v6" and bought one because it can be fully programmed, I have invested time in finding the best configuration for my needs through QMK Configurator and QMK Toolbox, however these tools do not allow me to do the following configuration of two keys specifically:

Expected Key 1 behavior:

In "layer 0", the key is set to change to "layer 1" by pressing and releasing quickly, now (once in layer 1) the same key has double function, "left ctrl" while holding down and switch to "layer 0" by pressing and releasing quickly

Code Key 1 behavior (according official documentation): 
 
    In layer 0 = TO (1)
    In layer 1 = LCTL_T (KC_TO (0))

Expected Key 2 behavior:

In "layer 0", the key is set to change to "layer 2" by pressing and releasing quickly, now (once in layer 2) the same key has double function, "left alt" while holding down and switch to "layer 0" by pressing and releasing quickly

Code Key 2 behavior (according official documentation): 
 
    In layer 0 = TO (2)
    In layer 2 = LALT_T (KC_TO (0))

In the case of the second key behavior I have tried to achieve it according to the official documentation trying other combinations like:

     In layer 0 = TO (0)
     In layer 1 = LALT_T (KC_TRANS)
------
     In layer 0 = KC_TRANS
     In layer 1 = LALT_T (KC_TRANS)
------
     In layer 0 = LALT_T (KC_TRANS)
     In layer 1 = LALT_T (KC_TRANS)
------
     ACTION_MODS_TAP_LAYER (MOD_LALT, 0)
------ 
     ACTION_MODS_TAP_KC (MOD_LALT, TO (0))
------
     ACTION_MOD_LALT_LAYER_ON (0)
------
    ACTION_MODS_TAP_TOGGLE (MOD_LALT, 0)
------
    ACTION_MOD_LALT_TOGGLE (0)

Unfortunately I have not been successful in achieving this configuration, also I do not know completely if this is possible since I am not a programmer, but if it is I ask you in the most attentive way could you please share information to obtain this result either in this way (or another ), I would really appreciate it too.

In advance, thank you very much, I share my actual configuration.

[image]

[image]

[image]

Myridium, 2021-10-04:
The [documentation](https://beta.docs.qmk.fm/using-qmk/advanced-keycodes/mod_tap) says that this can be achieved with

`MT(MOD_LCTL, TO(1))`

for your first example, and so on for the others.

However I do not recommend this approach as it is quite possibe to "tap" the key when you intended to hold it for a short period and release (e.g. in combination with a mouse click, or because you changed your mind about using a chordal combination of keys). If you insist, you may want to check [this page](https://beta.docs.qmk.fm/using-qmk/software-features/tap_hold) which details various intricacies with how the keyboard will decide about tapping/holding, and ways to customise it.

daskygit, 2021-10-04:
> `MT(MOD_LCTL, TO(1))`

This won't work as `MT` (this includes aliases like `LALT_T`) and `LT` only support basic keycodes, see [caveats](https://docs.qmk.fm/#/mod_tap?id=caveats).

To get functionality like above it will require some custom handling and compiling locally.

Myridium, 2021-10-04:
> > `MT(MOD_LCTL, TO(1))`
> 
> This won't work as `MT` (this includes aliases like `LALT_T`) and `LT` only support basic keycodes, see [caveats](https://docs.qmk.fm/#/mod_tap?id=caveats).
> 
> To get functionality like above it will require some custom handling and compiling locally.

What about a custom keycode that is processed in `process_record_user`?

daskygit, 2021-10-04:
Yes, but that's not going to help when using QMK Configurator.

senseimitz, 2021-10-04:
Thank you very much for the help Myridium

I have tried the command you shared and adjust it like this: 
     MT(MOD_LALT, MO(0)) 

However it did not work completely, the behavior is as follows:

    "Alt left" = when held (perfect)
     "m" = when pressing and releasing quickly (weird, do not switch to layer 0)

[image]

[image]

senseimitz, 2021-10-04:
> Yes, but that's not going to help when using QMK Configurator.

Thank you for you support daskygit, you're right but then, could you be so kind to share with me the correct way to do this setup please, I am very sorry if they are questions that seem absurd but I am not a programmer and it is difficult for me to understand the programming language

daskygit, 2021-10-04:
There's a few steps to go through before you start touching any code, so it's likely worth reading through the tutorial for setting up a [build environment](https://docs.qmk.fm/#/newbs_getting_started).  

Implementing some of the functionality you want isn't going to be trivial, especially as you will need some familiarity with C and the various QMK functions. I guess reading through the [tap dance documentation](https://docs.qmk.fm/#/feature_tap_dance?id=tap-dance-a-single-key-can-do-3-5-or-100-different-things) although it made be simpler to use something like [intercepting mod taps](https://docs.qmk.fm/#/mod_tap?id=intercepting-mod-taps) which includes examples. Either way is going to require some programming, no way around that.

senseimitz, 2021-10-05:
> There's a few steps to go through before you start touching any code, so it's likely worth reading through the tutorial for setting up a [build [...]

yes I understand perfectly that you say, I will gladly follow your advice and I'll review the documentation again, especially the topics it mentions, I hope I can understand what to know what to ask someone who knows how to program.

At the moment I am very happy to know the configuration I need to do is possible (although it is not easy) that was my biggest doubt.

Thank you very much in advance for your time and the support you have given me, this community is truly wonderful.

Have an excellent day

closed 2022-04-17 by stale[bot]