qmk/qmk_firmware #14781: [Feature Request] Send HID from usage page 0xFF01 (apple_vendor_keyboard)

adrienkohlbecker, 2021-10-10:
## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [x] New behavior

## Description

Hello,

In order to support the Spotlight key from new Apple keyboards, I would like to send usage ID `0x01` from page `0xFF01`, which is the `apple_vendor_keyboard` page.

You can find these values in Karabiner-Element's repository [here](https://github.com/pqrs-org/Karabiner-Elements/blob/fdc9d542a6f17258655f595e4d51d1e26aa25d41/src/vendor/cget/cget/pkg/pqrs-org__cpp-hid/install/include/pqrs/hid/usage_page.hpp#L60) and [here](https://github.com/pqrs-org/Karabiner-Elements/blob/173d7264e447dfb64d9be61f38805e86027d47fc/src/vendor/cget/cget/pkg/pqrs-org__cpp-hid/install/include/pqrs/hid/usage.hpp#L360).

I'm aware of `host_consumer_send`, and I'm not sure if `raw_hid_send` could help me and how to configure the `RAW_HID` feature, or if any other function already exists to do this. 🤔 

Any help is much appreciated

adrienkohlbecker, 2021-10-10:
Hey @fauxpark, thanks for replying. 

I'm already using the `fn` patch. With it and a product ID from Apple, I can get the old keyboard behavior (F4 matched to launchpad, F5 and F6 blanks). 

However, it doesn't look like using a product ID for a newer Apple Keyboard results in the correct behavior for the new F4-6 mappings. Which is why I'm trying to send those keys myself.

fauxpark, 2021-10-10:
It's not possible to do this from raw HID, the report descriptor is the wrong "shape".

What you could perhaps do is turn the reserved byte my patch repurposes just for the Fn usage into a bitfield, where each bit represents a usage for Spotlight, Fn, Mission Control, etc. Kind of like how the indicator LEDs are declared for reference, just in the opposite direction. I'll look into it myself when I get a chance.