qmk/qmk_firmware #14851: [Bug]

iraytrace, 2021-10-17:
Split keyboard only turns backlight per-key LEDs of on half that is directly connected to the computer when host system sleeps

## Describe the Bug
When the system sleeps, the half of the split keyboard not directly connected to the computer leaves the backlights on.

Quefrency keyboard Rev3 with latest QMK.  Host computer sleeps/powers down and the connected half of the keyboard turns off the per-key backlight LEDs.  On the other half of the keyboard the backlight LEDs stay lit.
## System Information

 - Keyboard:  
     QueFrency Rev3
   - Revision (if applicable):  
 - Operating system:
   Windows
 - AVR GCC version: 
  avr-gcc.exe (GCC) 8.4.0

 - ARM GCC version: 

 - QMK Firmware version:
0.14.23

 - Any keyboard related software installed? 
     - [X] AutoHotKey
     - [ ] Karabiner
     - [ ] Other:

## Additional Context

mb1986, 2021-11-28:
Can confirm the same behavior for Keebio Levinson rev3.

moroz-slavomir, 2021-12-15:
I have the same problem with kyria + elite-c.

stephaje, 2022-06-11:
Also seeing this problem on the keebio/iris/rev6.