qmk/qmk_firmware #14932: [Bug] Unnecessary delay when pressing the first key after a Mod-tap key

tomKPZ, 2021-10-24:
Mod-tap has 2 behaviors:

- When the mod-key is tapped, the following keypress is modified
- When the mod-key is held, all keypresses during the hold are modified

This leads to a necessary delay because if the modifier and then "a" and "b" are pressed in rapid succession, we don't yet know if the mod-key press is a tap or a hold, so we don't know whether to apply the modifier to "b" or not.

However, there the delay is unnecessary for "a".  In either case (tap or hold), "a" will be modified, so we can send the modified keypress right away.

drashna, 2021-10-24:
https://docs.qmk.fm/#/tap_hold

You probably want Ignore Mod Tap Interrupt

tomKPZ, 2021-10-24:
Thanks, I gave that a try but the behavior seems the same.  Also I need to correct a mistake in the bug report: I meant to say one-shot keys, not mod-tap keys.