qmk/qmk_firmware #14994: I can't make the firmware. There may be an error. I want to know how to modify it

1219008126, 2021-10-31:
This is the description that appeared when I made the firmware，I want to know what to do

Making matrixrgb with keymap via

avr-gcc.exe (GCC) 8.4.0
Copyright (C) 2018 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: keyboards/matrixrgb/matrixrgb.c                                                          [OK]
Compiling: keyboards/matrixrgb/keymaps/via/keymap.c                                                 [OK]
Compiling: quantum/quantum.c                                                                        [OK]
Compiling: quantum/send_string.c                                                                    [OK]
Compiling: quantum/bitwise.c                                                                        [OK]
Compiling: quantum/led.c                                                                            [OK]
Compiling: quantum/action.c                                                                         [OK]
Compiling: quantum/action_layer.c                                                                   [OK]
Compiling: quantum/action_macro.c                                                                   [OK]
Compiling: quantum/action_tapping.c                                                                 [OK]
Compiling: quantum/action_util.c                                                                    [OK]
Compiling: quantum/eeconfig.c                                                                       [OK]
Compiling: quantum/keyboard.c                                                                       [OK]
Compiling: quantum/keymap_common.c                                                                  [OK]
Compiling: quantum/keycode_config.c                                                                 [OK]
Compiling: quantum/logging/debug.c                                                                  [OK]
Compiling: quantum/logging/sendchar.c                                                               [OK]
Compiling: quantum/bootmagic/bootmagic_lite.c                                                       [OK]
Compiling: quantum/bootmagic/magic.c                                                                [OK]
Compiling: quantum/matrix_common.c                                                                  [OK]
Compiling: quantum/matrix.c                                                                         [OK]
Compiling: quantum/debounce/sym_defer_g.c                                                           [OK]
Compiling: quantum/main.c                                                                           [OK]
Compiling: quantum/color.c                                                                          [OK]
Compiling: quantum/rgb_matrix/rgb_matrix.c                                                          [OK]
Compiling: quantum/rgb_matrix/rgb_matrix_drivers.c                                                  [OK]
Compiling: lib/lib8tion/lib8tion.c                                                                  [OK]
Compiling: quantum/process_keycode/process_rgb.c                                                    [OK]
Compiling: platforms/avr/drivers/ws2812.c                                                           [OK]
Compiling: quantum/led_tables.c                                                                     [OK]
Compiling: quantum/via.c                                                                            [OK]
Compiling: quantum/dynamic_keymap.c                                                                 [OK]
Compiling: quantum/process_keycode/process_space_cadet.c                                            [OK]
Compiling: quantum/process_keycode/process_magic.c                                                  [OK]
Compiling: quantum/process_keycode/process_grave_esc.c                                              [OK]
Compiling: tmk_core/common/host.c                                                                   [OK]
Compiling: tmk_core/common/report.c                                                                 [OK]
Compiling: tmk_core/common/sync_timer.c                                                             [OK]
Compiling: tmk_core/common/usb_util.c                                                               [OK]
Compiling: tmk_core/common/avr/platform.c                                                           [OK]
Compiling: tmk_core/common/avr/suspend.c                                                            [OK]
Compiling: tmk_core/common/avr/timer.c                                                              [OK]
Compiling: tmk_core/common/avr/bootloader.c                                                         [OK]
Compiling: tmk_core/common/avr/printf.c                                                             [OK]
Compiling: tmk_core/protocol/lufa/lufa.c                                                            [OK]
Compiling: tmk_core/protocol/usb_descriptor.c                                                       [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Class/Common/HIDParser.c                                       [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Device_AVR8.c                                        [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/EndpointStream_AVR8.c                                [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Endpoint_AVR8.c                                      [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Host_AVR8.c                                          [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/PipeStream_AVR8.c                                    [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/Pipe_AVR8.c                                          [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/USBController_AVR8.c                                 [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/AVR8/USBInterrupt_AVR8.c                                  [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/ConfigDescriptors.c                                       [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/DeviceStandardReq.c                                       [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/Events.c                                                  [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/HostStandardReq.c                                         [OK]
Compiling: lib/lufa/LUFA/Drivers/USB/Core/USBTask.c                                                 [OK]
Compiling: tmk_core/protocol/lufa/usb_util.c                                                        [OK]
Linking: .build/matrixrgb_via.elf                                                                   [ERRORS]
 |
 | c:/qmk_msys/mingw64/bin/../lib/gcc/avr/8.4.0/../../../../avr/bin/ld.exe: C:\QMK_MSYS\tmp\ccUxtySM.ltrans0.ltrans.o: in function `RAINBOW_BEACON_math':
 | C:\Users\12190\qmk_firmware/quantum/rgb_matrix/animations/rainbow_beacon_anim.h:6: undefined reference to `g_led_config'
 | c:/qmk_msys/mingw64/bin/../lib/gcc/avr/8.4.0/../../../../avr/bin/ld.exe: C:\Users\12190\qmk_firmware/quantum/rgb_matrix/animations/rainbow_beacon_anim.h:6: undefined reference to `g_led_config'
 | c:/qmk_msys/mingw64/bin/../lib/gcc/avr/8.4.0/../../../../avr/bin/ld.exe: C:\Users\12190\qmk_firmware/quantum/rgb_matrix/animations/rainbow_beacon_anim.h:6: undefined reference to `g_led_config'
 | c:/qmk_msys/mingw64/bin/../lib/gcc/avr/8.4.0/../../../../avr/bin/ld.exe: C:\Users\12190\qmk_firmware/quantum/rgb_matrix/animations/rainbow_beacon_anim.h:6: undefined reference to `g_led_config'
 | c:/qmk_msys/mingw64/bin/../lib/gcc/avr/8.4.0/../../../../avr/bin/ld.exe: C:\QMK_MSYS\tmp\ccUxtySM.ltrans0.ltrans.o: in function `DUAL_BEACON_math':
 | C:\Users\12190\qmk_firmware/quantum/rgb_matrix/animations/dual_beacon_anim.h:6: undefined reference to `g_led_config'
 | c:/qmk_msys/mingw64/bin/../lib/gcc/avr/8.4.0/../../../../avr/bin/ld.exe: C:\QMK_MSYS\tmp\ccUxtySM.ltrans0.ltrans.o:C:\Users\12190\qmk_firmware/quantum/rgb_matrix/animations/dual_beacon_anim.h:6: more undefined references to `g_led_config' follow
 | collect2.exe: error: ld returned 1 exit status
 |
make[1]: *** [tmk_core/rules.mk:351：.build/matrixrgb_via.elf] 错误 1
Make finished with errors

1219008126, 2021-10-31:
This is the first time I use GitHub. Maybe it's not like this, but I don't understand it

fauxpark, 2021-10-31:
Looks like you've enabled RGB Matrix but haven't defined a `g_led_config`.

https://docs.qmk.fm/#/feature_rgb_matrix

1219008126, 2021-10-31:
I'm in keymap. c.  do  you want to add it to another folder

#include QMK_KEYBOARD_H

const uint16_t PROGMEM keymaps[][MATRIX_ROWS][MATRIX_COLS] = {

LAYOUT(
		KC_ESC, KC_1, KC_2, KC_3, KC_4, KC_5, KC_6, KC_7, KC_8, KC_9, KC_0, KC_MINS, KC_EQL, KC_BSPC, KC_INS,
		KC_TAB, KC_Q, KC_W, KC_E, KC_R, KC_T, KC_Y, KC_U, KC_I, KC_O, KC_P, KC_LBRC, KC_RBRC, KC_BSLS, KC_DEL,
		KC_CAPS, KC_A, KC_S, KC_D, KC_F, KC_G, KC_H, KC_J, KC_K, KC_L, KC_SCLN, KC_QUOT, KC_ENT, KC_PGUP,
		KC_LSFT, KC_Z, KC_X, KC_C, KC_V, KC_B, KC_N, KC_M, KC_COMM, KC_DOT, KC_SLSH, KC_RSFT, KC_UP, KC_PGDN,
		KC_LCTL, KC_LGUI, KC_LALT, KC_SPC, KC_RALT, MO(1), KC_LEFT, KC_DOWN, KC_RGHT
	),

LAYOUT(
		KC_GRV, KC_F1, KC_F2, KC_F3, KC_F4, KC_F5, KC_F6, KC_F7, KC_F8, KC_F9, KC_F10, KC_F11, KC_F12, KC_TRNS, KC_TRNS,
		KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_PSCR, KC_TRNS, KC_TRNS, RESET, KC_TRNS,
		KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_HOME,
		KC_TRNS, KC_SLEP, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_VOLU, KC_END,
		KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_TRNS, KC_MPLY, KC_VOLD, KC_MUTE
		)

};
#ifdef RGB_MATRIX_ENABL

led_config_t g_led_config = { {
  // Key Matrix to LED Index
  { 0, 1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14 },
  { 15, 16, 17, 18, 19, 20, 21, 22, 23, 24, 25, 26, 27, 28, 29 },
  { 30, 31, 32, 33, 34, 35, 36, 37, 38, 39, 40, 41, NO_LED, 42, 43 },
  { 44, 45, 46, 47, 48, 49, 50, 51, 52, 53, 54, 55, NO_LED, 56, 57 },
  { 58, 59, 60, 61, NO_LED, NO_LED, NO_LED, NO_LED, NO_LED, 62, 63, 64, NO_LED, 65, 66 }
}, {
  // LED Index to Physical Position
  { 0,0 }, { 16,0 }, { 32,0 }, { 48,0 }, { 64,0 }, { 80,0 }, { 96,0 }, { 112,0 }, { 128,0 }, { 144,0 }, { 160,0 }, { 176,0 }, { 192,0 }, { 208,0 }, { 224,0 },
  { 0,16 }, { 16,16 }, { 32,16 }, { 48,16 }, { 64,16 }, { 80,16 }, { 96,16 }, { 112,16 }, { 128,16 }, { 144,16 }, { 160,16 }, { 176,16 }, { 192,16 }, { 208,16 }, { 224,16 },
  { 0,32 }, { 16,32 }, { 32,32 }, { 48,32 }, { 64,32 }, { 80,32 }, { 96,32 }, { 112,32 }, { 128,32 }, { 144,32 }, { 160,32 }, { 176,32 }, { 192,32 }, { 208,32 },
	{ 0,48 }, { 16,48 }, { 32,48 }, { 48,48 }, { 64,48 }, { 80,48 }, { 96,48 }, { 112,48 }, { 128,48 }, { 144,48 }, { 160,48 }, { 176,48 }, { 192,48 }, { 208,48 },
	{ 0,64 }, { 16,64 }, { 32,64 }, { 96,64 }, { 160,64 }, { 176,64 }, { 192,64 }, { 208,64 }, { 224,64 }
}, {
  // LED Index to Flag
  1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1,
  1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1,
  1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1,
	1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1, 1,
  1, 1, 1, 1, 1, 1, 1, 1, 1,
} };

#endif

fauxpark, 2021-10-31:
The docs don't explicitly say what file it should go in, but it needs to be placed in your `<keyboard>.c`. Have you tried looking at other boards in the repo?

1219008126, 2021-10-31:
> The docs don't explicitly say what file it should go in, but it needs to be placed in your . Have you tried looking at other boards in the repo?`<keyboard>.c`
I tried to put it in < keyboard >. C, but it didn't seem to succeed. It's still the same error. I uploaded my file. I want to know where it is. Can you help me see what's wron
[matrixrgb.zip](https://github.com/qmk/qmk_firmware/files/7448347/matrixrgb.zip)
g

fauxpark, 2021-10-31:
[image]

fauxpark, 2021-10-31:
It seems like this code is from KBFirmware - that site uses a very old version of QMK, and we don't recommend using it anymore. You can convert the JSON file into more up to date code with this: https://noroadsleft.github.io/kbf_qmk_converter/

1219008126, 2021-10-31:
I just changed my file again and it still displays like this, so I think maybe this is not the proble
[matrix.zip](https://github.com/qmk/qmk_firmware/files/7448496/matrix.zip)
m

fauxpark, 2021-10-31:
See the image I posted.

1219008126, 2021-10-31:
I used the JSON file downloaded from the website kbfirmware.com, and used the content generated by the converter to overwrite the original content. I deleted #ifdef RGB_ MATRIX_ Enable, I don't know if that's right, but I don't quite understand

fauxpark, 2021-10-31:
> I deleted #ifdef RGB_ MATRIX_ Enable

No you didn't, you deleted `RGB_MATRIX_ENABL`...

1219008126, 2021-10-31:
Thank you very much for your help. I succeeded

closed 2021-11-01 by nooges