qmk/qmk_firmware #15045: STM32F401 Pin A9

jeromeli002, 2021-11-03:
The STM32F401 development board is used as a keyboard, and the A9 pin cannot be used. When A9 is used, it will always be pressed, and there will always be a key value output. I checked it online. This is that vbus needs to be modified and initialized for ordinary IO. How should I change it? ? And those feet can’t be used directly

drashna, 2021-11-03:
I believe that you cannot actually use the pin for the switch matrix or the like.  Along with A11, and A12. 

More details here: https://discord.com/channels/440868230475677696/440870965728116754/808224716102828042
but basically:

> Another quirk of F401 and F411 (not really specific to WeAct boards) is that, in additions to USB pins (A11 and A12), pins A9 and B2 are also not usable for the matrix (A9 has an undocumented internal pulldown due to its intended usage for USB VBUS sensing, and B2 is used as the BOOT1 input during reset, so it needs an external pulldown to make it possible to enter DFU). It may be possible to use pins A9 and B2 for some other purposes that are compatible with having a pulldown on the pin.

zvecr, 2021-11-04:
Duplicate of #7855

closed 2022-01-21 by zvecr