qmk/qmk_firmware #15081: KPRepublic BM65V2 addition request

rotaryking, 2021-11-06:
so im pretty retarded and can't figure out how to make my bm65rgb (rev 2) work since it's not directly supported within qmk, is it possible for it to be added soon

rotaryking, 2021-11-06:
idk if this will help but i grabbed these files from the kprepublic drive, contains json and hex files
[bm65v2.zip](https://github.com/qmk/qmk_firmware/files/7491164/bm65v2.zip)

sigprof, 2021-11-06:
There was some ongoing work to get that board supported: #15060 (apparently KP Republic does not want to do anything to get the board properly supported in QMK, therefore all the work has to be done by reverse engineering the existing PCB, which takes time and effort). Unfortunately, the person doing that work cannot continue it at the moment, because the PCB just died with no apparent reason, so the work is stalled until a PCB replacement can be obtained.

For the moment you can use the factory firmware (`bm65_via.hex`, and `bm65_via.json` to be loaded in VIA) published by KP Republic at https://drive.google.com/drive/folders/1KGApbq0qy5799h7ezkchcH7AxNTWbkW7 (that folder was restructured recently; if something like that happens again, and the link becomes invalid, try searching from https://drive.google.com/drive/folders/1W_gEDvrQ4U2cU7-NQ7ChMHBhihs1utPG to find the appropriate BM65V2 VIA firmware). Any questions about that firmware should be sent to KP Republic (but they don't even care to publish the sources of their QMK-based firmwares, as required by the GNU GPL license of QMK).

rotaryking, 2021-11-06:
i already have the provided files for the firmwares for via and hex. what i'm trying to do is to get it to work with 1000hz

peepeetee, 2021-11-10:
> 
> 
> i already have the provided files for the firmwares for via and hex. what i'm trying to do is to get it to work with 1000hz. my fault for not specifying that

I have a compiling copy of code in #15060 now but I can't test it, you can try it out if you feel adventurous enough

rotaryking, 2021-11-13:
im testing it right now and hoping my board doesnt get bricked

rotaryking, 2021-11-13:
it's flashed successfully and works properly. also, i made some changes to the files so it works at 1000hz.
only thing is now i cant use via to change keymaps, kill me now

peepeetee, 2021-11-22:
Added a via keymap in my PR

peepeetee, 2021-12-02:
Underglow was not working when you asked, now it's working. so if you want underglow just grab the latest commit

rotaryking, 2021-12-23:
i was too lazy to flash my keyboard to the newest until today and i have a few questions:
[image]
1. compiling error, is this supposed to happen or did i fuck something up
2. there's an extra keymaps folder that you made with one named "peepeetee" and another "via", are these worth looking at
3. managed to get via to work by myself a while ago

drashna, 2021-12-24:
The `__flash` needs to be changed to `PROGMEM`

rotaryking, 2021-12-24:
was resolved yesterday in a discord call with peepeetee

Banana-97, 2021-12-28:
@rotaryking May I ask how did you test the polling rate to see if it managed 1000hz? I have a KPrepublic BM80RBG, managed to flash with QMK 1000hz mod (following the link below) but I can't seem to get anything above 145hz (Default 125Hz)

[https://kevl.in/2021/02/1000hz-qmk-polling/]

rotaryking, 2021-12-28:
> @rotaryking May I ask how did you test the polling rate to see if it managed 1000hz? I have a KPrepublic BM80RGB, managed to flash with QMK 1000hz mod (following the link below) but I can't seem to get anything above 145hz (Default 125Hz)
> 
> [https://kevl.in/2021/02/1000hz-qmk-polling/]

According to peepeetee, the mcu that's used on the BM65RGB is only able to do 1000hz if the rgb is disabled. also, i used this:
```
#define FORCE_NKRO
#define USB_POLLING_INTERVAL_MS 1
// #define QMK_KEYS_PER_SCAN 67
```
on my board. Note that QMK KPS is disabled.

Banana-97, 2021-12-29:
> > @rotaryking May I ask how did you test the polling rate to see if it managed 1000hz? I have a KPrepublic BM80RGB, managed to flash with QMK 1000hz mod (following the link below) but I can't seem to get anything above 145hz (Default 125Hz)
> > [https://kevl.in/2021/02/1000hz-qmk-polling/]
> 
> According to peepeetee, the mcu that's used on the BM65RGB is only able to do 1000hz if the rgb is disabled. also, i used this:
> 
> ```
> #define FORCE_NKRO
> #define USB_POLLING_INTERVAL_MS 1
> // #define QMK_KEYS_PER_SCAN 67
> ```
> 
> on my board. Note that QMK KPS is disabled. Also, what are you using to check your kb polling rate?

I've added the define into the config, but still unable to get 1000 hz maybe it could also be the switches (Outemu Brown/Red) issues who knows, but the max I can get is 160hz

I use scan rate estimator in the link below to check the hz as I have a based line which is 500hz from Ducky one mini

[https://blog.seethis.link/scan-rate-estimator/](https://blog.seethis.link/scan-rate-estimator/)
[image]

sigprof, 2021-12-29:
> https://blog.seethis.link/scan-rate-estimator/

This test mostly measures the debounce delay; your 166 Hz result corresponds to 6 ms, which is close to the default 5 ms debounce time (extra 1 ms may come from the 1 ms polling interval, or just because you did not manage to get an exact 5 ms keypress).

Proper latency measurement would require some extra hardware (e.g., see https://michael.stapelberg.ch/posts/2021-05-08-keyboard-input-latency-qmk-kinesis/ — note that some controllers used in those tests support USB 2.0 High Speed and therefore can have 125 μs USB polling interval instead of 1 ms).

Banana-97, 2021-12-30:
> 

Oh ok, because I tested Ducky one mini and was able to get 500 Hz on that site so I was hoping my BM 80 will also be able to do it. Looks like the only way around this is to try out other debounce algorithm (not sure how) to get lower ms and I also bought new kailh silver speed switches to see if helps.. 

Yeah, the only way to properly check the latency extra hardware would definitely be needed.

clsmyth, 2022-01-16:
Hi @peepeetee .  I purchased a KPRepublic bm65rgb and had the misfortune of receiving one with the rev2 PCB.  After much searching I found this discussion.  I was able to cherry-pick the (currently) 20 commits from your "bm65rgbrev2" branch onto my branch in my QMK fork, add in my keymap, and successfully build and flash firmware onto my board.  Is there anything I can do to help your development process?

peepeetee, 2022-01-16:
> Hi @peepeetee . I purchased a KPRepublic bm65rgb and had the misfortune of receiving one with the rev2 PCB. After much searching I found this discussion. I was able to cherry-pick the (currently) 20 [...]

At this point the PR is ripe. We're just waiting for the collaborators to approve it

closed 2022-06-18 by github-actions[bot]

harmathy, 2022-12-04:
Please re-open! bm65rgb rev2 support is still missing.

peepeetee, 2022-12-05:
@harmathy oki poke me again in um a few days, I got the depression since um

let's take a look

since I last commented on #15060 which was in June

I should be able to work on this more to get it in, but it's finals season so if poke me in like 10 days

alternatively you can try it yourself if you can't wait the 9 days

harmathy, 2022-12-07:
@peepeetee With my poor knowledge, I managed to rebase your changes to 0.18.17 and get my custom layout to work on my keyboard. (https://codeberg.org/harmathy/qmk_firmware/)