qmk/qmk_firmware #15086: [Bug] LED Matrix not working with Bluetooth 4

myst729, 2021-11-07:
## Describe the Bug

Well here are my firmware files: https://github.com/myst729/qmk_firmware/tree/master/keyboards/dg/metax

It's a self-made testing board with custom key matrix (lite), IS31FL3731 driven LED matrix, and Bluetooth 4 (MDBT40 running Adafruit's firmware). The firmware can be successfully compiled when Bluetooth is disabled. It also can be compiled when LED matrix is disabled.

When both BLUETOOTH and LED MATRIX are enabled (in rules.mk), the compilation fails:
```
Compiling: tmk_core/protocol/lufa/adafruit_ble.cpp                                                 In file included from quantum/led_matrix/led_matrix.h:29,
                 from quantum/quantum.h:28,
                 from platforms/avr/drivers/analog.h:20,
                 from tmk_core/protocol/lufa/adafruit_ble.cpp:13:
drivers/led/issi/is31fl3731-simple.h:30:31: error: expected initializer before 'g_is31_leds'
 extern const is31_led __flash g_is31_leds[DRIVER_LED_TOTAL];
                               ^~~~~~~~~~~
 [ERRORS]
 | 
 | 
 | 
gmake[1]: *** [tmk_core/rules.mk:442: .build/obj_dg_metax_default/protocol/lufa/adafruit_ble.o] Error 1
gmake: *** [Makefile:510: dg/metax:default] Error 1
Make finished with errors
```

## System Information

 - Keyboard: Self made testing board
 - Operating system: macOS 12.0.1
 - AVR GCC version: 8.4.0
 - ARM GCC version: 8.3.1
 - QMK Firmware version: 0.14.23
 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

sigprof, 2021-11-07:
Looks like this combination of features hits a longstanding problem in GCC — named address spaces (e.g., `__flash`, which is used in the LED drivers to get much cleaner code than would be possible with `PROGMEM`) are supported only in C, but not in C++:

https://gcc.gnu.org/bugzilla/show_bug.cgi?id=69549

And the Bluetooth support is one of the rare parts of QMK that uses C++ (the other parts are the USB host code (used in the USB-USB converter) and some test code).

For this particular use case it could be possible to add some `#ifndef` guards to the offending part (the array in question is defined by the keyboard code, which should be in C, and used by the LED driver code; other parts of code should not need to access it):
```diff
diff --git a/drivers/led/issi/is31fl3731-simple.h b/drivers/led/issi/is31fl3731-simple.h
index ecde31eed5..09380f6922 100644
--- a/drivers/led/issi/is31fl3731-simple.h
+++ b/drivers/led/issi/is31fl3731-simple.h
@@ -27,7 +27,9 @@ typedef struct is31_led {
     uint8_t v;
 } __attribute__((packed)) is31_led;
 
+#ifndef __cplusplus  // __flash is not supported for C++
 extern const is31_led __flash g_is31_leds[DRIVER_LED_TOTAL];
+#endif
 
 void IS31FL3731_init(uint8_t addr);
 void IS31FL3731_write_register(uint8_t addr, uint8_t reg, uint8_t data);
```
(and the same fix will be needed for other LED matrix and RGB matrix drivers).

myst729, 2021-11-08:
The compilation passes but the firmware doesn't work. Guess I have to disable one of them at the moment.

drashna, 2021-11-09:
disable LTO

myst729, 2021-11-16:
Just find that TMK solves the issue like this: https://github.com/tmk/tmk_keyboard/commit/d14a00c44e6048e2126d6e14b4e849984674b168

closed 2022-01-29 by myst729