qmk/qmk_firmware #15100: [Bug] Inconsistent behaviour/unexpected outcome of setting MASTER_LEFT/RIGHT for splitting keyboard

TonyWu20, 2021-11-10:
Either one of "MASTER_LEFT/RIGHT" option will lead to no response from the slave side, and this is not consistent on my two dactyl manuform.

## Describe the Bug
I made two dactyl_manuform with pro micro as controllers. On the first build,  only setting "MASTER_LEFT" could let both hands work. Setting "MASTER_RIGHT" will make the left hand no response. However, on my second build, the situation reverses, only "MASTER_RIGHT" can make both hands work.

## System Information

 - Keyboard:  handwired/dactyl manuform
 - Operating system: macOS Big Sur 11.5.2
 - AVR GCC version:  8.4.0

 - ARM GCC version: 8.3.1 2.  20190703 (release) [gcc-8-branch revision 273027]

 - QMK Firmware version: 0.14.26

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

fuadsaud, 2021-11-11:
I'm experiencing a similar problem on my Ergodash Rev 1.2.

I have `MASTER_RIGHT` set due to my left side ProMicro's USB connector being broken. When flashing, only the right side works.

fuadsaud, 2021-11-11:
I did a bisect based on the latest version I knew to be working and the first bad commit appears to be ef92c9ee2cf4745637635ec1895399e4f013914c.

drashna, 2021-11-19:
Try adding `#define SPLIT_USB_DETECT` to your config.h, and see if that helps.

TonyWu20, 2021-11-19:
> Try adding `#define SPLIT_USB_DETECT` to your config.h, and see if that helps.

I tried adding
```
#define SPLIT_USB_DETECT
#define SPLIT_USB_TIMEOUT 2000
```
but it didn't help

drashna, 2021-12-15:
> I have `MASTER_RIGHT` set due to my left side ProMicro's USB connector being broken. When flashing, only the right side works.

That may be something completely different. And ... that you may need to reflash both sides.  ISP flashing the pro micro may be a good way to go, in this case. 

> > Try adding `#define SPLIT_USB_DETECT` to your config.h, and see if that helps.
> 
> I tried adding
> 
> ```
> #define SPLIT_USB_DETECT
> #define SPLIT_USB_TIMEOUT 2000
> ```
> 
> but it didn't help

Which pro micro's are these?

TonyWu20, 2021-12-19:
> 
> > > Try adding `#define SPLIT_USB_DETECT` to your config.h, and see if that helps.
> > 
> > 
> > I tried adding
> > ```
> > #define SPLIT_USB_DETECT
> > #define SPLIT_USB_TIMEOUT 2000
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > but it didn't help
> 
> Which pro micro's are these?
I tried this on my second keyboard's pro micros.

daskygit, 2022-10-06:
I'm unable to reproduce this locally, issues like this tend to be caused by faulty vbus detection. Although adding `#define SPLIT_USB_DETECT` and flashing both sides should resolve this.

Closing the issue as stale, If you're able to replicate this with recent a firmware version you're welcome to reopen.

closed 2022-10-06 by daskygit