qmk/qmk_firmware #15112: [Feature Request] Support Durgod K320 Corona (with white LEDs)

wadimw, 2021-11-10:
Support for Durgod K320 Corona (white LED backlight) on STM32F070RBT6.

## Feature Request Type

- [X] Core functionality
- [X] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [ ] Alteration (enhancement/optimization) of existing feature(s)
- [ ] New behavior

## Description

@dkjer @tylert

I currently own the keyboard in question and would like to help adding support for it (including the backlight). My hardware revision is the same chip as currently supported base, so I suspect that there's not that much work to be done, but it surely would be easier to collaborate with someone who already has experience with this hardware rather than starting from scratch.

dkjer, 2021-11-10:
@wadimw Do you have ANSI or ISO layout?  I have an ANSI Corona somewhere, and I have an experimental branch with backlight support that I could try to resurrect

wadimw, 2021-11-10:
Mine is ANSI as well

wadimw, 2021-12-14:
@dkjer anything I can do to help?

closed 2022-06-18 by github-actions[bot]

lucaam, 2023-12-13:
Hello @dkjer,

Any plan to introduce this feature? Any help needed? I own the keyboard and I could test or help introduce this feature but I am not able to do it myself.