qmk/qmk_firmware #15120: [Feature Request] Unicode support for non-qwerty layouts?

trougnouf, 2021-11-11:
## Feature Request Type

- [ ] Core functionality
- [ ] Add-on hardware support (eg. audio, RGB, OLED screen, etc.)
- [x] Alteration (enhancement/optimization) of existing feature(s)
- [x] New behavior

## Description

It doesn't seem possible to use the unicode functions with an alternative keyboard layout, because (from what I understand) the UC function (and others defined in https://beta.docs.qmk.fm/using-qmk/software-features/feature_unicode ) expect a hexadecimal code which is then typed as qwerty with the OS-specific activator. If the necessary code (eg "20AC" for the euro sign) falls outside of the hexadecimal range in the OS layout (eg "20AI" on dvorak), then there is no way to input it.

It would be very useful if unicodes could be typed in as strings, or could be somehow combined with the keymap_extras header files (s.a. keymap_dvp.h).

drashna, 2021-11-19:
It should already support this.  Try adding `#include "sendstring_dvp.h"` to your keymap.c

trougnouf, 2021-11-22:
That seems to work :) Thank you @drashna 
Somehow I still don't get the unicode character with UC(0x20AC) though, something weird happens in the terminal (no time to catch it) and Firefox goes two tabs to the left ... I captured the wev output, any idea what could be the issue?
```
[14:     wl_keyboard] key: serial: 459633; time: 1299850428; key: 36; state: 0 (released)
                      sym: Return       (65293), utf8: ''
[14:     wl_keyboard] key: serial: 459634; time: 1299852828; key: 64; state: 1 (pressed)
                      sym: Alt_L        (65513), utf8: ''
[14:     wl_keyboard] modifiers: serial: 0; group: 16
                      depressed: 00000008: Mod1
                      latched: 00000000
                      locked: 00000010: Mod2
[14:     wl_keyboard] key: serial: 459636; time: 1299852836; key: 50; state: 1 (pressed)
                      sym: Shift_L      (65505), utf8: ''
[14:     wl_keyboard] modifiers: serial: 0; group: 16
                      depressed: 00000009: Shift Mod1
                      latched: 00000000
                      locked: 00000010: Mod2
[14:     wl_keyboard] key: serial: 459638; time: 1299852844; key: 17; state: 1 (pressed)
                      sym: 2            (50), utf8: '2'
[14:     wl_keyboard] key: serial: 459639; time: 1299852852; key: 17; state: 0 (released)
                      sym: 2            (50), utf8: ''
[14:     wl_keyboard] key: serial: 459640; time: 1299852860; key: 50; state: 0 (released)
                      sym: Shift_L      (65505), utf8: ''
[14:     wl_keyboard] modifiers: serial: 0; group: 16
                      depressed: 00000008: Mod1
                      latched: 00000000
                      locked: 00000010: Mod2
[14:     wl_keyboard] key: serial: 459642; time: 1299852868; key: 50; state: 1 (pressed)
                      sym: Shift_L      (65505), utf8: ''
[14:     wl_keyboard] modifiers: serial: 0; group: 16
                      depressed: 00000009: Shift Mod1
                      latched: 00000000
                      locked: 00000010: Mod2
[14:     wl_keyboard] key: serial: 459644; time: 1299852876; key: 16; state: 1 (pressed)
                      sym: 0            (48), utf8: '0'
[14:     wl_keyboard] key: serial: 459645; time: 1299852884; key: 16; state: 0 (released)
                      sym: 0            (48), utf8: ''
[14:     wl_keyboard] key: serial: 459646; time: 1299852892; key: 50; state: 0 (released)
                      sym: Shift_L      (65505), utf8: ''
[14:     wl_keyboard] modifiers: serial: 0; group: 16
                      depressed: 00000008: Mod1
                      latched: 00000000
                      locked: 00000010: Mod2
[14:     wl_keyboard] key: serial: 459648; time: 1299852900; key: 38; state: 1 (pressed)
                      sym: a            (97), utf8: 'a'
[14:     wl_keyboard] key: serial: 459649; time: 1299852908; key: 38; state: 0 (released)
                      sym: a            (97), utf8: ''
[14:     wl_keyboard] key: serial: 459650; time: 1299852916; key: 31; state: 1 (pressed)
                      sym: c            (99), utf8: 'c'
[14:     wl_keyboard] key: serial: 459651; time: 1299852924; key: 31; state: 0 (released)
                      sym: c            (99), utf8: ''
[14:     wl_keyboard] key: serial: 459652; time: 1299852932; key: 64; state: 0 (released)
                      sym: Alt_L        (65513), utf8: ''
[14:     wl_keyboard] modifiers: serial: 0; group: 16
                      depressed: 00000000
                      latched: 00000000
                      locked: 00000010: Mod2
```
Desired keys are LCTRL+LSHIFT+U 20AC

drashna, 2021-12-13:
you may need/want to replace the unicode input start and finish functions, as that will allow you to replace the existing functionality, and fix the issues.

https://docs.qmk.fm/#/feature_unicode?id=start-and-finish-input-functions

https://github.com/qmk/qmk_firmware/blob/f718a10889e6adf33f3fc2f41b61cad7fe9e0c2e/quantum/process_keycode/process_unicode_common.c#L81-L169

closed 2022-06-18 by github-actions[bot]