qmk/qmk_firmware #15125: [Bug] ergodox_stm32_default.elf  | arm-none-eabi-gcc: error: .build/obj_ergodox_stm32_default/lib/printf/printf.o: No such file or directory

mianqi2016, 2021-11-11:
## Describe the Bug
I used "qmk compile" for ergodox_stm32, there was such error prompt:
https://ibb.co/WcdPNd6

Linking: .build/ergodox_stm32_default.elf                                                           [ERRORS]
 | 
 | arm-none-eabi-gcc: error: .build/obj_ergodox_stm32_default/lib/printf/printf.o: No such file or directory
 | 
make[1]: *** [tmk_core/rules.mk:351: .build/ergodox_stm32_default.elf] Error 1

## System Information

 - Keyboard:  ergodox_stm32
   - Revision (if applicable):  
 - Operating system: Ubuntu 20.04
 - AVR GCC version: 

 - ARM GCC version: 
 - arm-none-eabi-gcc (15:9-2019-q4-0ubuntu1) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]

 - QMK Firmware version:
 - qmk --version
1.0.0

git describe --abbrev=0 --tags
0.14.16

 - Any keyboard related software installed? 
   - [ ] AutoHotKey
   - [ ] Karabiner
   - [ ] Other:

## Additional Context

drashna, 2021-11-19:
Compiles fine for me.  You're probably missing submodules. 

Run `make git-submodule` from the qmk_firmware directory and recompile.

mianqi2016, 2021-12-02:
Since the net was too slow, so I download .zip from qmk@github, while:
`qmk compile -kb ergodox_stm32 -km default
Ψ Compiling keymap with make --jobs=1 ergodox_stm32:default

⚠ "git describe --abbrev=6 --dirty --always --tags" returned error code 128
`

`make git-submodule
make: *** No rule to make target 'git-submodule'.  Stop.
`

fauxpark, 2021-12-02:
Downloading the zip file from GitHub won't work, you need to follow the setup instructions.

mianqi2016, 2021-12-07:
Now, there is a new error:
`Linking: .build/ergodox_stm32_default.elf                                                           [ERRORS]
 | 
 | arm-none-eabi-gcc: error: .build/obj_ergodox_stm32_default/lib/printf/printf.o: No such file or directory
 | 
make[1]: *** [tmk_core/rules.mk:365: .build/ergodox_stm32_default.elf] Error 1
Make finished with errors
`

drashna, 2021-12-13:
`qmk clean -a` and/or `make git-submodule` again.

mianqi2016, 2021-12-14:
It still was there:
arm-none-eabi-gcc: error: .build/obj_ergodox_stm32_default/lib/printf/printf.o: No such file or directory

and I saw in the folder".build/obj_ergodox_stm32_default/lib/", there was only "chibios", no "printf".

fauxpark, 2021-12-14:
What does `git status` show?

mianqi2016, 2021-12-15:
The output for "git status":
`On branch master
Your branch is up to date with 'origin/master'.

Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
  (commit or discard the untracked or modified content in submodules)
	modified:   ../../../lib/printf (modified content)
	modified:   ../../../lib/vusb (modified content)

no changes added to commit (use "git add" and/or "git commit -a")
`

mianqi2016, 2021-12-15:
The output of "ls lib" under "qmk_firmware":

`
arm_atsam  chibios-contrib  lib8tion  printf  usbhost
chibios    googletest       lufa      python  vusb

`

fauxpark, 2021-12-15:
Delete the `lib/printf` and `lib/vusb` folders, then run `make git-submodule` again.

mianqi2016, 2021-12-16:
Now, it seems OK:
`
qmk compile -kb ergodox_stm32 -km default
Ψ Compiling keymap with make --jobs=1 ergodox_stm32:default

QMK Firmware 0.15.6
Making ergodox_stm32 with keymap default

arm-none-eabi-gcc (15:9-2019-q4-0ubuntu1) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
Copyright (C) 2019 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

Compiling: lib/printf/printf.c                                                                      [OK]
Linking: .build/ergodox_stm32_default.elf                                                           [OK]
Creating binary load file for flashing: .build/ergodox_stm32_default.bin                            [OK]
Creating load file for flashing: .build/ergodox_stm32_default.hex                                   [OK]

Size after:
   text	   data	    bss	    dec	    hex	filename
      0	  46194	      0	  46194	   b472	.build/ergodox_stm32_default.hex

Copying ergodox_stm32_default.bin to qmk_firmware folder                                            [OK]
(Firmware size check does not yet support STM32F103; skipping)

`
While, what does the last sentence "(Firmware size check does not yet support STM32F103; skipping)" mean?

fauxpark, 2021-12-16:
You can ignore that, it just means you won't get the % of flash space used by the firmware at the end of compilation.

closed 2021-12-24 by drashna