qmk/qmk_firmware #15134: [Bug] OSM SHIFT sometimes behaves incorrectly (oryx, moonlander)

urza, 2021-11-12:
One of thumb keys on my Moonlander [is configured](https://configure.zsa.io/moonlander/layouts/4JADQ/latest/0) as **one shot shift modifier** (and nothing else).
Occasionally it doesn't produce shifted next letter but regular lowercase letter.
The problem is not in timing. This happens when the osm shift key is pressed and released withing the tapping term so should produce shifted character.

I created simple debugging app that prints keyUp and keyDown events and prints them with timestamp (second.milliseconds).
On screenshot below you can see that few times osm shift worked as expected, but then one time it produced lower character because keyboard sent:
```
shift down
shift up
X down
shift up
X up
```

The first shift up seems like bug?

[image]

## System Information

 - Keyboard:  
   Moonlander mk I, configured with oryx online, flashed with Wally
 - Operating system:
  Windows 10
 
 - Any keyboard related software installed? 
   No

urza, 2021-11-13:
I did some more testing and it seems that something strange is going on. I work mostly remote via RDP (remote desktop from windows 10  to another windows 10), this is where the osm shift is sometimes failing, I would say the rate of failing to produce osm shift is 1 in 10. It happens when the osm shift key is pressed very swiftly for as short period as possible between key down and key up. 
However on my local computer, the failure doesn't seem to be happening.

I didn't think RDP could play a role since the keyboard should send key events to local OS and that should be just forwarded to remote OS as is.
But googling this and it seems that this may indeed be RDP issue: 
https://superuser.com/questions/219461/shift-and-control-keys-out-of-sync-with-normal-keys-over-rdp

I wonder if anyone encountered similar issue with RDP?

urza, 2021-11-13:
Related issue https://github.com/qmk/qmk_firmware/issues/1037

drashna, 2021-11-19:
Yup, that's a known issue. And there isn't a simple fix for it, IIRC

TartCodes, 2023-03-30:
> I did some more testing and it seems that something strange is going on. I work mostly remote via RDP (remote desktop from windows 10 to another windows 10), this is where the osm shift is sometimes [...]

Hi there!

not sure you will see this but I am having this same issue when I RDP. It;s kinda of a killer because I bought this KB for work purposes, but the layer almost never works and is really ruining my productivity - has there been a fix or a workaround?